KiCad/kicad-winbuilder #1: Does not work on initial setup

nickoe, 2015-03-25:
Trying to use this today, and I could not make it succeed in actually start building. Is this what the currently latest commit (ea7f1320f4dd2dec435776cafd91f14a912c39b7) is referring as currently broken?

	make_i686.bat
	[...SNIP...]
	downloading guile-2.0.11-3-i686.pkg.tar.xz...
	downloading guile-2.0.11-3-i686.pkg.tar.xz...
	downloading guile-2.0.11-3-i686.pkg.tar.xz...
	downloading guile-2.0.11-3-i686.pkg.tar.xz...
	downloading guile-2.0.11-3-i686.pkg.tar.xz...
	downloading make-4.1-3-i686.pkg.tar.xz...
	downloading make-4.1-3-i686.pkg.tar.xz...
	downloading make-4.1-3-i686.pkg.tar.xz...
	downloading make-4.1-3-i686.pkg.tar.xz...
	downloading make-4.1-3-i686.pkg.tar.xz...
	error: failed to commit transaction (unexpected error)
	Errors occurred, no packages were upgraded.
	-- Running pacman --noconfirm -Su
	:: Starting full system upgrade...
	 there is nothing to do
	-- HOME_DIR C:/Users/johndoe/KiCad-Winbuilder/msys32/home/johndoe
	-- KICAD_PACKAGE_SOURCE_DIR C:/Users/johndoe/KiCad-Winbuilder/msys32/home/noste
	rga/MINGW-packages/mingw-w64-kicad-git/
	-- Running cd ~ %% git clone https://github.com/Alexpux/MINGW-packages.git
	-- Running cd ~/MINGW-packages/mingw-w64-kicad-git && export CARCH=i686 && makep
	kg-mingw -s --noconfirm
	/usr/bin/bash: line 0: cd: /home/johndoe/MINGW-packages/mingw-w64-kicad-git: No
	 such file or directory
	Press any key to continue . . .

BrianSidebotham, 2015-03-27:
Sorry I missed this Nick.

The problem is not present locally for me, I don't know why I'm out of
sync, I'll look tonight.

The problem lies in the command cd ~ %% git clone
https://github.com/Alexpux/MINGW-packages.git

It's a simple typo and should be cd ~ && git clone
https://github.com/Alexpux/MINGW-packages.git

BrianSidebotham, 2015-03-27:
Also, with regards to the "currently broken" comment, this is due to the
fact that no package is built for me currently. It's a strange problem. The
whole of KiCad is built, but for some reason pacman installs KiCad rather
than packages it. I end up with a package file that's only a couple of k
long and KiCad installed in the msys2 environment. I'm unsure when things
stopped working and started behaving like this.

I could do with some help solving that problem!

nickoe, 2015-03-27:
Ok, great. I will try again when I get home.

I did not use much time to debug the error hence I reported the issue
instead. Great that it was just a typo. :P

closed 2015-03-27 by BrianSidebotham

ciccioswing, 2015-10-12:
Good morning.

I have the same problem, but I don't understand how did you workaround. Could you help me please? I have downloaded the last kicad-winbuilder (and there aren't any "%%" in the .cmake file but just "&&"). These are the last few lines before the error:

Compiling /mingw64/lib/python2.7/site-packages...
-- Running pacman --noconfirm -Su
:: Aggiornamento del sistema in corso...
 non ci sono aggiornamenti disponibili
-- HOME_DIR
-- KICAD_PACKAGE_SOURCE_DIR /MINGW-packages/mingw-w64-kicad-git/
-- MSYS2 user name is: $USERNAME=
-- Running cd /MINGW-packages/mingw-w64-kicad-git &&  makepkg-mingw -s --noconfirm
2>&1: riga 0: cd: /MINGW-packages/mingw-w64-kicad-git: No such file or directory

Do you see something strange? Please let me know if you have any suggestion.

Kind regards,

Francesco

commit f1cee18f (2015-03-27): Fix a typo (Fixes #1) and direct stderr to logs too