KiCad/kicad-winbuilder #7: How to Build KiCad

bvernoux, 2015-10-06:
A "How to Build KiCad" will be welcome and especially to find where the final build is ...
Anyway you have done an amazing work on this builder ! (I was previously using https://launchpad.net/kicad-winbuilder)

nickoe, 2015-10-06:
Run one of the bat files, please feel free to test. It has actually been a long time since I used those directly, usually I just run makepkg-mingw and the copydlls.sh script.

bvernoux, 2015-10-06:
I have tested with make_x86_64.bat (using trunk SHA-1: 709a6b2c4c22c93880d4b9916b2377ca2d663d39)
Running bat file work to download & install msys and all dependencies.
I can see the build start and all seems fine except for the last steps to finalize the binaries ... (and the .bat windows just close).
In order to fix this I have done following steps:
* Double click on KiCad-Winbuilder\msys64\msys2_shell.bat
* cd to KiCad-Winbuilder (root)
* cmake -Dx86_64=ON -P KiCad-Winbuilder.cmake

And all is fine without any error and final binaries with installer are in
KiCad-Winbuilder\msys64\home\UserName\out\pack-x86_64\
With lot of directories and kicad-product-r6248.20a3e23-x86_64.exe installer.

nickoe, 2015-10-08:
Have you tried to execute `make_x86_64.bat` from the normal `cmd.exe` prompt? I am not sure that what you are trying to do is supposed to work.

bvernoux, 2015-10-08:
Double clicking on .bat or launching it from cmd is the same (as the .bat will be executed in the same cmd environment), but I have not tested it, as it was solved using a real "linux" shell by using KiCad-Winbuilder\msys64\msys2_shell.bat which is really better than normal windows console which does not support color ...

nickoe, 2015-10-09:
Yes indeed. But as according to the issue title, feel free tp create a pull request with description on how to use. I am a bit busy for the time being.

FranzForstmayr, 2015-10-09:
I think i've triggered the same error now, but i don't know how to __cd to KiCad Winbuilder__, because the directory is above the __msys2_shell.bat__. Can you please explain what exactly you are doing?

bvernoux, 2015-10-10:
Alternative way to build Kicad is clearly defined on official site to
http://kicad-pcb.org/contribute/build-windows/

FranzForstmayr, 2015-10-12:
Thank you, 
I did not find this site before, maybe it should be linked to the dropdown menu of "Contribute" on the KiCad main page.

nickoe, 2015-11-20:
Can we close this issue or?

bvernoux, 2015-11-20:
Could you updated the https://github.com/nickoe/KiCad-Winbuilder/blob/master/readme.md before to close ? (updating with a procedure with a how to a bit like it is done in this issue ?)

bevanweiss, 2016-01-30:
There is a bit more of a 'how to' of building with the WinBuilder now.  I think this issue could be closed.

closed 2016-02-08 by nickoe