KiCad/kicad-winbuilder #8: Error in script "install.nsi" makensis.exe

kinichiro, 2015-10-08:
Hi,
I'm using KiCad-Winbuilder on Windows 7 (64 bit) (Japanese).

When winbuilder was creating the package, 
makensis.exe was terminated with this error.
<pre>
...
Processing script file: "install.nsi" (ACP)
Bad text encoding: install.nsi:9
Error in script "install.nsi" on line 9 -- aborting creation process
</pre>

The line 9 of nsis/install.nsi is copyright comment line like this.
`; Copyright (C) 2015 Nick Østergaard`

I replaced the character "Ø" with "O", then error had gone away.
I'm afraid that makensis.exe is not able to handle the script that contains multi byte char.

My original LANG environment variable is `ja_JP.UTF-8`.
I tried to use `en_US.UTF-8` and `da_DK.UTF-8`, but I got the same error.

Has anyone else encountered this issue, too ?

nickoe, 2015-10-08:
Hmm, that is strange. Maybe it si the language enviroment of the windows install itself. I guess mine would be en_US.UTF-8, but I am not at my usual machine so I can test. But then the Jenkins is building on a german locale.

nickoe, 2016-09-18:
Does this problem still happen to you @kinichiro?

kinichiro, 2016-09-20:
@nickoe Recently I had not experienced this. I would like to close this.

closed 2016-09-20 by kinichiro