KiCad/kicad-winbuilder #9: Building Local Version

FranzForstmayr, 2015-10-08:
Hey, i got the kicad sources from the git mirror and want to try several things with the sources locally.
Is it possible to get Kicad-Winbuilder to build a local repository?
I'm working on Win7x64.

nickoe, 2015-10-08:
Yes it should of course be possible. In theory you should be able to use cmake with -G"MSYS Makefiles" to build manually inside the MSYS2 environment. But it did not work a few months ago, but did before that time. Maybe it is fixed now. But that is from the MSYS2 side of things.

FranzForstmayr, 2015-10-08:
Hey,
i pushed my git repo to *\Kicad-Winbuilder\msys64\git\KiCad and changed the source in the BUILD file to 
__source=("${_realname}"::"git+file:///git/KiCad/.git"
"${_realname}-i18n"::"git+https://github.com/KiCad/kicad-i18n.git"
"${_realname}-libs"::"git+https://github.com/KiCad/kicad-library.git"__
Now it seems to get the right files.

nickoe, 2015-10-08:
Yes that should also work, but will not make fast recompiles because it starts from the beginning, but it should work.

closed 2015-10-08 by nickoe