KiCad/kicad-winbuilder #11: Build errors

xarx00, 2015-10-30:
During a completely new build I met with several problems:

1) The build ended In KiCad-Winbuilder.cmake, l.323. That was because at l. 314, the directory "${CMAKE_SOURCE_DIR}/${MSYS2}/home" was empty. That's why the value HOME_DIR was empty too, and "cd ${HOME_DIR}" at l.323 failed. I fixed this by changing the command at l.314 to:
```
    file( GLOB HOME_DIR "${CMAKE_SOURCE_DIR}/${MSYS2}/home*" )
```
(removing the last slash)

2) After the previous fix, the build went on, but ended during the build of KiCad itself with the error
```
.....
[ 29%] [32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_other/tune_diff_pair_skew_legend.cpp.obj[0m
[ 29%] [32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_other/tune_single_track_length_legend.cpp.obj[0m
[ 29%] [32m[1mLinking CXX static library libbitmaps.a[0m
[ 29%] Built target bitmaps
Makefile:127: recipe for target 'all' failed
```
as a result of several previous targets having failed, e.g.
```
common/CMakeFiles/pcb_lexer_source_files.dir/build.make:69: recipe for target 'V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/msys64/home/MINGW-packages/mingw-w64-kicad-git/src/kicad/common/pcb_keyword

s.cpp' failed
CMakeFiles/Makefile2:546: recipe for target 'common/CMakeFiles/pcb_lexer_source_files.dir/all' failed
```
The pcb_keywords.cpp is missing in the repository, moreover it is listed in the .gitignore file.

I don't understand how and when the make-files are generated, so I wasn't able to fix the second problem.

I can provide log files for both problems, but I was not able to attach them to the issue.

Best regards,
     Martin.

nickoe, 2015-10-30:
How exactly did you execute the build?

xarx00, 2015-10-30:
I started it from the cmd by executing make_x86_64.bat.

nickoe, 2015-10-30:
The windows command prompt, cmd.exe?

xarx00, 2015-10-30:
Yes.

nickoe, 2015-10-30:
Before you do too much debugging, I suggest you to try from scratch again. Just create a new clone of the kicad-winbuilder repo, such that you keep the old logs to compare the new agianst.

I user issue I think I have seen, but I am not exactly sure why it might happen.

The make files are generated by cmake in the PKGBUILD (insude the msys2/mingw environment) that is run by the cmake from the bat file.

xarx00, 2015-10-30:
Can I ask you... Should "${CMAKE_SOURCE_DIR}/${MSYS2}/home" be empty at the time the make gets to KiCad-Winbuilder.cmake, l.323? If not, what should be inside?

nickoe, 2015-10-30:
There should be a directory with your windows username.

What does the this line result in? https://github.com/nickoe/KiCad-Winbuilder/blob/master/KiCad-Winbuilder.cmake#L319

xarx00, 2015-10-30:
And to the second problem... Should the various *_keywords.cpp files (like the pkb_keywords.cpp mentioned above) be part of the KiCad build or not? I.e. were they put into .gitignore by mistake or not?

xarx00, 2015-10-30:
The username is empty there.

nickoe, 2015-10-31:
The keywords cpp files are probably generated by swig or something. I think the issue is only that home path.

I do not understand why there is no directory in the home dir. I will not be able to debug it any more right now. Try to just create a dir in the home dir and rerun. But as said earlier, try a clean run before debugging too much.

xarx00, 2015-10-31:
Keyword.cpps... the problem is that the build tries to compile them, even though they are not there. A PKGBUILD file must list them explicitly. For instance, pcb_keywords.cpp is listed in "...\msys64\home\MINGW-packages\mingw-w64-kicad-git\src\kicad\common\CMakeLists.txt".

xarx00, 2015-10-31:
I started a clean build today while creating the "..../home/username" directory manually. And after three or four re-runs, it went on successfully. Unfortunately it failed at what seems to be the very end:
```
==> Creating package "mingw-w64-x86_64-kicad-git"...
  -> Generating .PKGINFO file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Finished making: mingw-w64-x86_64-kicad-git r6288.e64c36a-1 (Sat, Oct 31, 20
15  9:20:48 PM)
←[1m←[33m=> WARNING:←(B←[m←[1m You don't have installed mingw-w64 toolchain for
architecture i686.←(B←[m
←[1m←[33m=> WARNING:←(B←[m←[1m To install it run: 'pacman -S mingw-w64-i686-tool
chain'←(B←[m
-- Running ~/copydlls.sh                          --arch=x86_64
         --pkgpath=$HOME/MINGW-packages/mingw-w64-kicad-git
     --nsispath=$HOME/nsis                          --makensis=V:/Martin/BuiltAp
ps/KiCad/kicad-winbuilder/trunk/.support/nsis-3.0b1/Bin/makensis.exe
2>&1: /c/Users/xarx/copydlls.sh: No such file or directory
```
Please note that the build was run from "V:/KiCad-Winbuilder" (where the "copydlls.sh" is), not from "C:/Users" (where it is not). 
And that the error says something about the i686 architecture, while I ran "make_x86_64.bat".

 I also re-run the build from yesterday, and it went to the same point too. So my second problem from yesterday has resolved by itself, perhaps someone committed a fix to the KiCad repository. But still no overall success.

What would you suggest to me to do now?

Thank you,
 Martin

xarx00, 2015-10-31:
I've found the last problem mentioned above: In "KiCad-Winbuilder.cmake" at l.349 and l.363 there should be "${HOME_DIR}" everywhere instead of "~" and "$HOME". Because "$HOME" points to "/c/Users/xarx".

So there were two problems in all:
1) The home/username dir was not created
3) $HOME instead of ${HOME_DIR}

However, even after the last fix, the "/c/Users/xarx" dir is still mentioned in the log. Is this correct? (I cannot attach the log, so I'm copying its content here:

```
-- HOME_DIR V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/msys64/home
-- KICAD_PACKAGE_SOURCE_DIR V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/msys64/home/MINGW-packages/mingw-w64-kicad-git/
-- MSYS2 user name is: $USERNAME=home
-- Running V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/msys64/home/copydlls.sh                          --arch=x86_64                          --pkgpath=V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/msys64/home/MINGW-packages/mingw-w64-kicad-git                          --nsispath=V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/msys64/home/nsis                          --makensis=V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/.support/nsis-3.0b1/Bin/makensis.exe$ARCH=x86_64
$PKGPATH=V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/msys64/home/MINGW-packages/mingw-w64-kicad-git
$NSISPATH=V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/msys64/home/nsis
$MAKENSIS=V:/Martin/BuiltApps/KiCad/kicad-winbuilder/trunk/.support/nsis-3.0b1/Bin/makensis.exe
warning: using hardcoded outdir path
Decoded pkg is x86_64 and r6288.e64c36a
64bit

x86_64 x86_64
$TARGETDIR=/c/Users/xarx/out/pack-x86_64
$MSYSDIR=/mingw64
Output will be in /c/Users/xarx/out/pack-x86_64
Copying dll depends...
*wx*.dll
*glew*.dll
*jpeg*.dll
libcairo*.dll
*ssl*.dll
libgomp*.dll
libstd*.dll
libgcc*.dll
libwinpthread-1.dll
libboost*.dll
libeay32.dll
ssleay32.dll
libpng*.dll
libpixman*.dll
libfreetype*.dll
libfontconfig*.dll
libharfbuzz*.dll
libexpat*.dll
libbz2*.dll
libglib*.dll
libiconv*.dll
zlib*.dll
libintl*.dll
libtiff*.dll
liblzma*.dll
libpython*.dll
libxml*.dll
libxslt*.dll
libexslt*.dll
xsltproc.exe
Copying include/python2.7...
Copying lib/python2.7...
Copying python.exe...
Building NSIS insaller exe...
/c/Users/xarx
======================
/c/Users/xarx/out/pack-x86_64/nsis
Processing config: V:\Martin\BuiltApps\KiCad\kicad-winbuilder\trunk\.support\nsis-3.0b1\nsisconf.nsh
Processing script file: "install.nsi" (ACP)
warning: File: "..\AUTHORS.txt" -> no files found. (install.nsi:186)
warning: File: "..\license_for_documentation.txt" -> no files found. (install.nsi:188)
warning: File: "..\share\doc\kicad\tutorials\*" -> no files found. (install.nsi:222)
warning: !verbose: Pop failed, stack is empty (macro:MUI_UNGETLANGUAGE:2)
warning: !verbose: Pop failed, stack is empty (macro:MUI_UNGETLANGUAGE:24)

Processed 1 file, writing output (x86-ansi):

Output: "C:\Users\xarx\out\pack-x86_64\kicad-product-r6288.e64c36a-x86_64.exe"
Install: 5 pages (320 bytes), 8 sections (2 required) (16576 bytes), 17391 instructions (486948 bytes), 7356 strings (181408 bytes), 1 language table (394 bytes).
Uninstall: 1 page (128 bytes), 1 section (2072 bytes), 137 instructions (3836 bytes), 133 strings (2457 bytes), 1 language table (242 bytes).
Datablock optimizer saved 87341 KiB (~27.2%).

Using zlib compression.

EXE header size:               39936 / 35840 bytes
Install code:                 140810 / 677894 bytes
Install data:              238585490 / 1070627407 bytes
Uninstall code+data:           15354 / 17660 bytes
CRC (0xA7CE4797):                  4 / 4 bytes

Total size:                238781594 / 1071358805 bytes (22.2%)

5 warnings:
  File: "..\AUTHORS.txt" -> no files found. (install.nsi:186)
  File: "..\license_for_documentation.txt" -> no files found. (install.nsi:188)
  File: "..\share\doc\kicad\tutorials\*" -> no files found. (install.nsi:222)
  !verbose: Pop failed, stack is empty (macro:MUI_UNGETLANGUAGE:2)
  !verbose: Pop failed, stack is empty (macro:MUI_UNGETLANGUAGE:24)
/c/Users/xarx
```

xarx00, 2016-02-07:
I returned to KiCad after some time, made a clean build, and had to solve the exactly same problems as before:
* Under "${CMAKE_SOURCE_DIR}/${MSYS2}/home", there is no directory created automatically
* "~" does not refer to that home directory, but to my Windows home directory instead.

Here is the patch:

```
--- KiCad-Winbuilder.cmake	Fri Feb 05 11:58:13 2016
+++ KiCad-Winbuilder.cmake	Sun Feb 07 08:50:09 2016
@@ -312,6 +312,12 @@
 # KiCad pacman package source
 # Get the home directory
 file( GLOB HOME_DIR "${CMAKE_SOURCE_DIR}/${MSYS2}/home/*" )
+if ( NOT HOME_DIR )
+    execute_msys2_bash( "whoami" "${LOG_DIR}/whoami" )
+	file(STRINGS "${LOG_DIR}/whoami" USERNAME )
+	file( MAKE_DIRECTORY "${CMAKE_SOURCE_DIR}/${MSYS2}/home/${USERNAME}" )
+	file( GLOB HOME_DIR "${CMAKE_SOURCE_DIR}/${MSYS2}/home/*" )
+endif()
 set( KICAD_PACKAGE_SOURCE_DIR "${HOME_DIR}/MINGW-packages/mingw-w64-kicad-git/" )
 message( STATUS "HOME_DIR ${HOME_DIR}" )
 message( STATUS "KICAD_PACKAGE_SOURCE_DIR ${KICAD_PACKAGE_SOURCE_DIR}" )
@@ -346,10 +352,10 @@
           DESTINATION "${HOME_DIR}" )
 
     # Copy the runtime requirements (shared objects mainly)
-    execute_msys2_bash( "$HOME/copydlls.sh \
+    execute_msys2_bash( "${HOME_DIR}/copydlls.sh \
                          --arch=i686 \
-                         --pkgpath=$HOME/MINGW-packages/mingw-w64-kicad-git \
-                         --nsispath=$HOME/nsis \
+                         --pkgpath=${HOME_DIR}/MINGW-packages/mingw-w64-kicad-git \
+                         --nsispath=${HOME_DIR}/nsis \
                          --makensis=${NSIS_MAKE_COMMAND}"
                          "${LOG_DIR}/copydlls_mingw32" )
 endif()
@@ -360,10 +366,10 @@
           DESTINATION "${HOME_DIR}" )
 
     # Copy the runtime requirements (shared objects mainly)
-    execute_msys2_bash( "~/copydlls.sh \
+    execute_msys2_bash( "${HOME_DIR}/copydlls.sh \
                          --arch=x86_64 \
-                         --pkgpath=\$HOME/MINGW-packages/mingw-w64-kicad-git \
-                         --nsispath=$HOME/nsis \
+                         --pkgpath=\${HOME_DIR}/MINGW-packages/mingw-w64-kicad-git \
+                         --nsispath=${HOME_DIR}/nsis \
                          --makensis=${NSIS_MAKE_COMMAND}"
                          "${LOG_DIR}/copydlls_mingw64" )
 endif()

```

Mamonbo, 2016-03-06:
I have two things to ask you:
+ Do you have files in `\share\doc\kicad\tutorials\` finally?
+ Does your username in Windows have multibyte characters?

nickoe, 2017-12-22:
Some changes have happened since this, so I will close this for now. Feel free to open a new ticket if it still fails for you.

closed 2017-12-22 by nickoe