KiCad/kicad-winbuilder #16: windows 10 th2 not compatible

marekr, 2015-12-09:
Just putting this here as documentation.

Windows 10 TH2 update broke forking a 64-bit process from a 32-bit process. 
For some reason the winbuilder system is broken when trying to use it. MSYS2 standalone works just fine.

cygwin which msys2 uses did have the bug themselves and theres a "test" release to fix it but I don't think its cygwin as manual msys2 building works.

https://cygwin.com/ml/cygwin/2015-12/msg00003.html

bevanweiss, 2016-01-02:
I believe this has now been fixed upstream (and is available in the latest msys2-runtime package).

It appears that the Winbuilder itself isn't properly updating the msys install using pacman during the process (although it does look like it should be in there).  In the meantime the process is...
1. Launch msys2_shell.bat
2. pacman -Sy
3. pacman --needed -S bash pacman pacman-mirrors msys2-runtime
4. exit (shell should close)
5. Re-launch msys2_shell.bat
6. pacman -Su
7. exit (shell should close)

Then running the make_all.bat should work.

I'll see if I can fix up the inbuilt pacman issue and then submit a pull request.  There's a few other minor fixups and updates that I've made.

nickoe, 2016-01-02:
FWIW, the MSYS2 people have made a new script to help updating the msys2-runtime, it is called `update-core`. After that is run, you need to respawn the shell, this replaces step 3.

bevanweiss, 2016-01-02:
I haven't had any luck getting the `update-core` to actually work.  It's always just quietly finished, without actually updating anything the few times that I've tried it so far.

I had added it to the pull request, but since it's been a bit flakey for me I also left in the original version (both didn't seem to hurt).

nickoe, 2016-02-08:
Was this the same issue as mentioned here? https://lists.launchpad.net/kicad-developers/msg22944.html

bevanweiss, 2016-02-08:
I don't believe so.  The boost issue mentioned (via link) was a runtime issue I believe.
This github issue was purely related to development environment, and the MSYS2 build environment.

I believe this issue should be closed as it's not really a bug inside Winbuilder, but just with MSYS2, and it's been fixed in the latest release.

marekr, 2016-02-08:
To correct, its not a MSYS2 bug but rather a CYGWIN issue which MSYS2 uses. BUT the way we use winbuilder more or less caused the bug as well, there were hackish ways to avoid it.

closed 2016-02-08 by marekr