KiCad/kicad-winbuilder #17: Build process stops on applying the patch

milzec, 2015-12-10:
During the last 3 days I am trying to complete the build process but it fails.
Last attempts I made today on Windows 7 Professional 64-bit, clean build (from the start), by starting make_all.bat from command prompt. The last lines in command prompt window are:

==> Extracting sources...
  -> Creating working copy of kicad-source-mirror git repo...
Switched to a new branch 'makepkg'
  -> Creating working copy of kicad-i18n git repo...
Switched to a new branch 'makepkg'
  -> Creating working copy of kicad-library git repo...
Switched to a new branch 'makepkg'
  -> Extracting kicad-doc-HEAD.tar.gz with bsdtar
==> Starting pkgver()...
==> Updated version: mingw-w64-x86_64-kicad-git r6364.eaef5c1-1
==> Removing existing $pkgdir/ directory...
==> Starting build()...
  -> Patching...
fatal: can't open patch '../../0001-Replace-avhttp-with-libcurl-implementation.patch': No
such file or directory
==> ERROR: A failure occurred in build().
    Aborting...

C:\kc>
-----------------

I've checked for the 0001-Replace-avhttp-with-libcurl-implementation.patch file and it is there; didn't know how to check if directory is correct or not.

Kind regards,
Miloje

nickoe, 2015-12-10:
I don't beleive you are on the latest version.

milzec, 2015-12-10:
I downloaded KiCad-Winbuilder-master.zip file today (3 hours ago). I've just checked the file  copydlls.sh and it contains the line_173:   "libffi*.dll" )  .  It looks like the latest one.
For the last two weeks didn't have this problem, but I was using the zip file downloaded in October.
On another laptop (Win8.1 Home 64-bit) I started to get the error (but I have to check if it is the same) even if I didn't change the build package to the latest version.
Everything started like 3 days ago.

Regards

nickoe, 2015-12-10:
It looks strange. It is not supposed to use the PKGBUILD's that is located in the repo root.

milzec, 2015-12-10:
Hi Nick,
Let me know if I should provide some more details on this (I'm relatively new to this KiCad builder).

Regards

nickoe, 2015-12-10:
Ahh yearh, right. I do indeed copy the PKGBUILD from the root.

nickoe, 2015-12-10:
You could just revert to using the version before https://github.com/nickoe/KiCad-Winbuilder/commit/e7d90b1fdd979090401abe985b0a2487d5c02c15

But you might need to install the glm for both archs.

milzec, 2015-12-10:
Hi Nick,

Thanks for the info.

Best regards,
Miloje

nickoe, 2015-12-12:
It should be fixed now, please test.

closed 2015-12-12 by nickoe

milzec, 2015-12-14:
Working as expected. Thank you.

Best regards,
Miloje