KiCad/kicad-winbuilder #18: Windows installer: help files

ElektroQuark, 2015-12-14:
Hi,

I'm a KiCad for Windows user.
I use my own libraries, so when installing KiCad I deselect both Schematic
and Footprint libraries. Required space for installation goes from 1.1GB to
366.3MB. Fine.
But if I deselect the help files it goes to 177.3MB. So more than half of
the installation are the help files.
Can the installer be changed to select the help files language and bypass
the rest of the files?

Keep the good work!
Thanks!

Iñigo.

P.S. I know storage space is cheap. But I really hate unneeded files.

nickoe, 2015-12-14:
I sure understand you. The documentation files are quite big. This is a low priority, I might get the urge to fix it at some time, but I have other KiCad related tasks that I would like to fix first.

bevanweiss, 2016-01-30:
As of the commit Nick did a few weeks ago, this should now be done.  You can individually select to install/not install the various language help files.
I believe this issue could be closed now..

closed 2016-02-08 by nickoe