KiCad/kicad-winbuilder #25: Could not build KiCad on a fresh Ki-WB instalce - PGP keys issues

keruseykaryu, 2016-01-30:
I'm trying to build KiCad with fresh KiCad-Winbuilder instance under Windows 7 HP x64 and Cmake 3.4.3. First issues I'm getting with PGP keys (e-mail addresses scrambled manually):

```
gpg: /etc/pacman.d/gnupg/trustdb.gpg: trustdb created
gpg: no ultimately trusted keys found
gpg: Generating pacman keyring master key...
gpg: skipping control `%no-protection' ()
..+++++
+++++
gpg: key F578ED20 marked as ultimately trusted
gpg: Done
==> Updating trust database...
gpg: 3 marginal(s) needed, 1 complete(s) needed, PGP trust model
gpg: depth: 0  valid:   1  signed:   0  trust: 0-, 0q, 0n, 0m, 0f, 1u
==> Appending keys from msys2.gpg...
gpg: 3 marginal(s) needed, 1 complete(s) needed, PGP trust model
gpg: depth: 0  valid:   1  signed:   0  trust: 0-, 0q, 0n, 0m, 0f, 1u
==> Locally signing trusted keys in keyring...
  -> Locally signing key D55E7A6D7CE9BA1587C0ACACF40D263ECA25678A...
  -> Locally signing key 123D4D51A1793859C2BE916BBBE514E53E0D0813...
  -> Locally signing key B91BCF3303284BF90CC043CA9F418C233E652008...
  -> Locally signing key 9DD0D4217D75A33B896159E6DA7EF2ABAEEA755C...
==> Importing owner trust values...
gpg: setting ownertrust to 4
gpg: setting ownertrust to 4
gpg: setting ownertrust to 4
gpg: inserting ownertrust of 4
==> Updating trust database...
gpg: 3 marginal(s) needed, 1 complete(s) needed, PGP trust model
gpg: depth: 0  valid:   1  signed:   4  trust: 0-, 0q, 0n, 0m, 0f, 1u
gpg: depth: 1  valid:   4  signed:   3  trust: 0-, 0q, 0n, 4m, 0f, 0u
gpg: depth: 2  valid:   3  signed:   0  trust: 3-, 0q, 0n, 0m, 0f, 0u
gpg: refreshing 8 keys from hkp://pool.sks-keyservers.net
gpg: requesting key F578ED20 from hkp server pool.sks-keyservers.net
gpg: requesting key CA25678A from hkp server pool.sks-keyservers.net
gpg: requesting key AEEA755C from hkp server pool.sks-keyservers.net
gpg: requesting key 3E0D0813 from hkp server pool.sks-keyservers.net
gpg: requesting key 3E652008 from hkp server pool.sks-keyservers.net
gpg: requesting key A47D45A1 from hkp server pool.sks-keyservers.net
gpg: requesting key 2C51581E from hkp server pool.sks-keyservers.net
gpg: requesting key 4CA56930 from hkp server pool.sks-keyservers.net
gpgkeys: key DA5070B0FE01E59408E1CF7A79C2455FF578ED20 not found on keyserver
gpg: DBG: armor-keys-failed (KEY 0xDA5070B0FE01E59408E1CF7A79C2455FF578ED20 BEGI
N
) ->0
gpg: DBG: armor-keys-failed (KEY 0xDA5070B0FE01E59408E1CF7A79C2455FF578ED20 FAIL
ED 6
) ->6
gpg: key CA25678A: "Alexey ... (..) <alexey.xxxxxx@xxxxx.xxx>" 4 new signatures
gpg: key AEEA755C: "Martell ... (..) <martelxxxxxxx@xxxxx.xxx>" 3 new signatures
gpg: key 3E0D0813: "Ray ... (MSYS2 Developer - master key) <mingw.xxxxxxx@xxxxx.xxx>" 2 new signatures
gpg: key 3E652008: "Ignacio Casal ... <icquinxxxxx@xxxxx.xxx>" 2 new signatures
gpg: key A47D45A1: "Alexey ... (..) <alexxxx@xxxxx.xxx>" 1 new signature
gpg: key 2C51581E: "Martell ... (MSYS2 Developer) <martellxxxxxx@xxxxx.xxx>" not changed
gpg: key 4CA56930: "Ray ... (MSYS2 Developer) <mingw.xxxxxxx@xxxxx.xxx>" 2 new signatures
gpg: Total number processed: 7
gpg:              unchanged: 1
gpg:         new signatures: 14
gpg: keyserver communications error: key not found
gpg: keyserver communications error: bad public key
gpg: 3 marginal(s) needed, 1 complete(s) needed, PGP trust model
gpg: depth: 0  valid:   1  signed:   4  trust: 0-, 0q, 0n, 0m, 0f, 1u
gpg: depth: 1  valid:   4  signed:   3  trust: 0-, 0q, 0n, 4m, 0f, 0u
gpg: depth: 2  valid:   3  signed:   0  trust: 3-, 0q, 0n, 0m, 0f, 0u
gpg: keyserver refresh failed: bad public key
==> ERROR: A specified local key could not be updated from a keyserver.
```

This issues causes a plenty of package corruption messages (translated to English):

```
(...)
:: File /var/cache/pacman/pkg/mingw-w64-x86_64-cmake-3.4.1-1-any.pkg.tar.xz is corrupted (Ivalid or damaged package (PGP signature)).
Do you wish to delete it? [Y/n] 
:: File /var/cache/pacman/pkg/mingw-w64-x86_64-wineditline-2.101-4-any.pkg.tar.xz is corrupted (Ivalid or damaged package (PGP signature)).
Do you wish to delete it? [Y/n] 
:: File /var/cache/pacman/pkg/mingw-w64-x86_64-pcre-8.37-2-any.pkg.tar.xz is corrupted (Ivalid or damaged package (PGP signature)).
Do you wish to delete it? [Y/n] 
:: File /var/cache/pacman/pkg/mingw-w64-x86_64-swig-3.0.6-1-any.pkg.tar.xz jestuszkodzony (Ivalid or damaged package (PGP signature)).
Do you wish to delete it? [Y/n] 
:: File /var/cache/pacman/pkg/mingw-w64-x86_64-libxml2-2.9.2-6-any.pkg.tar.xz is corrupted (Ivalid or damaged package (PGP signature)).
Do you wish to delete it? [Y/n] 
:: File /var/cache/pacman/pkg/mingw-w64-x86_64-libgpg-error-1.20-1-any.pkg.tar.xz is corrupted (Ivalid or damaged package (PGP signature)).
Do you wish to delete it? [Y/n] 
:: File /var/cache/pacman/pkg/mingw-w64-x86_64-libgcrypt-1.6.4-1-any.pkg.tar.xz is corrupted (Ivalid or damaged package (PGP signature)).
Do you wish to delete it? [Y/n] 
:: File /var/cache/pacman/pkg/mingw-w64-x86_64-libxslt-1.1.28-5-any.pkg.tar.xz is corrupted (Ivalid or damaged package (PGP signature)).
Do you wish to delete it? [Y/n]
Errors occured, no packages has been updated.
-- Running pacman --noconfirm -Su
:: Starting full system update
 Nothing to do
-- HOME_DIR E:/KiCad-Winbuilder/msys64/home/Kerusey
-- KICAD_PACKAGE_SOURCE_DIR E:/KiCad-Winbuilder/msys64/home/Kerusey/MINGW-packages/mingw-w64-kicad-git/
-- MSYS2 user name is: $USERNAME=Kerusey
-- Running cd "E:/KiCad-Winbuilder/msys64/home/Kerusey" && git clone https://github.com/Alexpux/MINGW-packages.git
2>&1: git: command not found
-- Running cd "E:/KiCad-Winbuilder/msys64/home/Kerusey/MINGW-packages/mingw-w64-kicad-git" &&  makepkg-mingw -s --noconfirm
 WARNING: You don't have installed mingw-w64 toolchain for architecture x86_64.
 WARNING: To install it run: 'pacman -S mingw-w64-x86_64-toolchain'
 WARNING: You don't have installed mingw-w64 toolchain for architecture i686.
 WARNING: To install it run: 'pacman -S mingw-w64-i686-toolchain'
```

I was trying few times to start from a scratch this week - same result.

xarx00, 2016-02-07:
I had the same problem a few days ago, but it has solved by itself. Probably someone fixed the keys. I suggest to make a **clean** build again, it might be OK already.

keruseykaryu, 2016-02-07:
The build is in progress now, but `gpg` still reports same key issues. Packages has been updated anyway.

nickoe, 2016-03-02:
@keruseykaryu Are you still struggeling with this issue?

keruseykaryu, 2016-03-03:
The package corruption reports and broken toolchain warnings are gone. The gpg issues still exist, but the whole build script passes and finally installer is created. However, last week I wanted to run a newly compiled KiCad and none of his applications did not want to run at all. I have to start build again. Maybe it's just a temporary problem.

nickoe, 2016-03-03:
I think it was just a hickup in the toolchain. But just to make sure you are aware, there are also nightlies on the download page, if all you do it rebuilt to get some newer version.

closed 2016-03-03 by nickoe