KiCad/kicad-winbuilder #26: copydlls.sh fails during new build

xarx00, 2016-02-06:
During a clean build, copydlls.sh fails (on Win7pro x64):

```
-- Running V:/workspaces/KiCad/winbuilder/msys64/home/copydlls.sh                          --arch=x86_64                          --pkgpath=V:/workspaces/KiCad/
winbuilder/msys64/home/MINGW-packages/mingw-w64-kicad-git                          --nsispath=V:/workspaces/KiCad/winbuilder/msys64/home/nsis
       --makensis=V:/workspaces/KiCad/winbuilder/.support/nsis-3.0b3/Bin/makensis.exe
$ARCH=x86_64
$PKGPATH=V:/workspaces/KiCad/winbuilder/msys64/home/MINGW-packages/mingw-w64-kicad-git
$NSISPATH=V:/workspaces/KiCad/winbuilder/msys64/home/nsis
$MAKENSIS=V:/workspaces/KiCad/winbuilder/.support/nsis-3.0b3/Bin/makensis.exe
warning: using hardcoded outdir path
Decoded pkg is x86_64 and r6538.7c637ea
64bit
x86_64 x86_64
$TARGETDIR=/c/Users/mplechsmid/out/pack-x86_64
$MSYSDIR=/mingw64
Output will be in /c/Users/mplechsmid/out/pack-x86_64
Copying dll depends...
*wx*.dll
cp: missing file operand
Try 'cp --help' for more information.
*glew*.dll
cp: missing file operand
Try 'cp --help' for more information.
*jpeg*.dll
cp: missing file operand
Try 'cp --help' for more information.
.....

```
This is because /msys64/usr/bin/find.exe does not work. And that is because early in the build process the following fails:

```
Packages (4) bash-4.3.042-4  msys2-runtime-2.4.1.16860.40c26fc-1  pacman-5.0.0.6348.cc5a8f1-1  pacman-mirrors-20160112-1

Total Download Size:   10.93 MiB
Total Installed Size:  51.44 MiB
Net Upgrade Size:       8.44 MiB

:: Proceed with installation? [Y/n]
:: Retrieving packages ...
downloading msys2-runtime-2.4.1.16860.40c26fc-1-x86_64.pkg.tar.xz...
downloading msys2-runtime-2.4.1.16860.40c26fc-1-x86_64.pkg.tar.xz...
downloading msys2-runtime-2.4.1.16860.40c26fc-1-x86_64.pkg.tar.xz...
downloading msys2-runtime-2.4.1.16860.40c26fc-1-x86_64.pkg.tar.xz...
downloading msys2-runtime-2.4.1.16860.40c26fc-1-x86_64.pkg.tar.xz...
downloading bash-4.3.042-4-x86_64.pkg.tar.xz...
downloading bash-4.3.042-4-x86_64.pkg.tar.xz...
downloading bash-4.3.042-4-x86_64.pkg.tar.xz...
downloading bash-4.3.042-4-x86_64.pkg.tar.xz...
downloading bash-4.3.042-4-x86_64.pkg.tar.xz...
downloading pacman-mirrors-20160112-1-any.pkg.tar.xz...
downloading pacman-mirrors-20160112-1-any.pkg.tar.xz...
downloading pacman-mirrors-20160112-1-any.pkg.tar.xz...
downloading pacman-mirrors-20160112-1-any.pkg.tar.xz...
downloading pacman-mirrors-20160112-1-any.pkg.tar.xz...
downloading pacman-5.0.0.6348.cc5a8f1-1-x86_64.pkg.tar.xz...
downloading pacman-5.0.0.6348.cc5a8f1-1-x86_64.pkg.tar.xz...
downloading pacman-5.0.0.6348.cc5a8f1-1-x86_64.pkg.tar.xz...
downloading pacman-5.0.0.6348.cc5a8f1-1-x86_64.pkg.tar.xz...
downloading pacman-5.0.0.6348.cc5a8f1-1-x86_64.pkg.tar.xz...
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
checking available disk space...
upgrading msys2-runtime...
error: cannot remove /usr/bin/msys-2.0.dll (Permission denied)
warning: warning given when extracting /usr/bin/msys-2.0.dll (Can't unlink already-existing object)
upgrading bash...
error: cannot remove /usr/bin/bash.exe (Permission denied)
warning: warning given when extracting /usr/bin/bash.exe (Can't unlink already-existing object)
upgrading pacman-mirrors...
upgrading pacman...
error: cannot remove /usr/bin/pacman.exe (Permission denied)
warning: warning given when extracting /usr/bin/pacman.exe (Can't unlink already-existing object)
```
It's because msys-2.0.dll, bash.exe and pacman.exe are being used at that time.
After I replace msys-2.0.dll manually, find.exe starts to work. 

So the problem is evidently in that the new find.exe is no longer backwards-compatible with the old binary msys-2.0.dll distributed as part of msys2-base-x86_64-20150916

xarx00, 2016-02-07:
There is an even worse consequence of the problem above: The tar archive mingw-w64-x86_64-kicad-git-r6540.4c7a61d-1-any.pkg.tar.xz in /msys64/home/MINGW-packages/mingw-w64-kicad-git/ is generated corrupted. After fixing the above, the build finishes correctly.

nickoe, 2016-09-18:
I will close this as it seems to be a transient hiccup with MSYS2.

closed 2016-09-18 by nickoe