KiCad/kicad-winbuilder #27: Uninstaller behaviour

ElektroQuark, 2016-02-09:
KiCad uninstaller doesn't delete Start menu entries.

nickoe, 2016-02-09:
Which exact entries?

ElektroQuark, 2016-02-09:
Hi Nick,

KiCad installer version: kicad-product-r6543.5349eb4-i686

Just installed:
[image]

After uninstall:
[image]

nickoe, 2016-02-09:
Please try the next nightly build.

nickoe, 2016-02-10:
@ElektroQuark please try kicad-product-r6546.a319c34 from the nightlies.

ElektroQuark, 2016-02-10:
@nickoe, I will try tomorrow and will post results here.

ElektroQuark, 2016-02-11:
It's solved. Last nightly build works OK.
Thanks Nick.

closed 2016-02-11 by ElektroQuark

nickoe, 2016-02-11:
Thank your for reporting :)