KiCad/kicad-winbuilder #28: Missing python2w.exe

xarx00, 2016-02-22:
Lately in the KiCad developers' mailing list (https://lists.launchpad.net/kicad-developers/msg23255.html), there were expressed wishes to include python2w.exe in the Windows build. Today, I ran into the same problem. Actually, the fix is simple, I'm attaching the patch:

[copydlls.sh.patch.zip](https://github.com/nickoe/KiCad-Winbuilder/files/140384/copydlls.sh.patch.zip)

nickoe, 2016-02-22:
I can commit this as is, but what you really should do it to make a github pull request, then the patch will be attributed to you properly. I guess you are new to github, so if you want consider this an exorcise. See https://help.github.com/articles/creating-a-pull-request/

nickoe, 2016-02-22:
Did you also test if the exe worked or not?

xarx00, 2016-02-22:
I thought I wouldn't have the privileges to commit anything here. But I'll try that.

I performed a simple wxpython Hello World example, and that worked. Both from command-line and from KiCad Spice netlist exporter.

nickoe, 2016-02-22:
@xarx00 The method on github, is that you fork the repo to your own account and upload your patch to that in a branch and on the web interface you create the pull request, and I can then accept or reject it.

closed 2016-03-02 by nickoe