KiCad/kicad-winbuilder #33: Build failing

mikewilt, 2016-03-05:
I'm back to this after being away for a while.  This is a new build on a machine that hasn't done it before.  The first hint of trouble looks like this:

[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/export_options_pad.cpp.obj←[0m
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/export.cpp.obj←[0m
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/fabrication.cpp.obj←[0m
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/file_footprint.cpp.obj←[0m
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/fill_zone.cpp.obj←[0m
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/find.cpp.obj←[0m
[ 11%] ←[32m←[1mLinking CXX static library libpolygon.a←[0m
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/find_replace.cpp.obj←[0m
C:/Users/Mike/Documents/kicad-winbuilder-master/msys64/mingw32/bin/ranlib.exe: libpolygon.a: Permission denied
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/flag.cpp.obj←[0m
make[2]: *** [polygon/libpolygon.a] Error 1
polygon/CMakeFiles/polygon.dir/build.make:172: recipe for target 'polygon/libpolygon.a' failed
make[2]: *** Deleting file 'polygon/libpolygon.a'
CMakeFiles/Makefile2:1767: recipe for target 'polygon/CMakeFiles/polygon.dir/all' failed
make[1]: *** [polygon/CMakeFiles/polygon.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/footprint_text.cpp.obj←[0m
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/fonts.cpp.obj←[0m
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/gbr_select_mode0.cpp.obj←[0m

Then if I try to run make_all.bat again it just does this:

C:\Users\Mike\Documents\kicad-winbuilder-master>make_all
-- HOME_DIR C:/Users/Mike/Documents/kicad-winbuilder-master/msys64/home/Mike
-- KICAD_PACKAGE_SOURCE_DIR C:/Users/Mike/Documents/kicad-winbuilder-master/msys64/home/Mike/MINGW-packages/mingw-w64-kicad-git/
-- MSYS2 user name is: $USERNAME=Mike
-- Running cd "C:/Users/Mike/Documents/kicad-winbuilder-master/msys64/home/Mike/MINGW-packages/mingw-w64-kicad-git" &&  makepkg-mingw -s --noconfirm
==> Making package: mingw-w64-x86_64-kicad-git r6494.edc47c3-1 (Sat, Mar 05, 2016  9:50:43 AM)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating kicad git repo...
Fetching origin
From https://github.com/KiCad/kicad-source-mirror
 + 08d0c81...d72ba0f refs/pull/5/merge -> refs/pull/5/merge  (forced update)
  -> Updating kicad-i18n git repo...
Fetching origin
From https://github.com/KiCad/kicad-i18n
   62fed86..59adf27  master     -> master
  -> Updating kicad-libs git repo...
Fetching origin
From https://github.com/KiCad/kicad-library
 x [deleted]         (none)     -> refs/pull/384/merge
 x [deleted]         (none)     -> refs/pull/385/merge
 x [deleted]         (none)     -> refs/pull/414/merge
 x [deleted]         (none)     -> refs/pull/432/merge
 x [deleted]         (none)     -> refs/pull/433/merge
   e571896..02bca80  master     -> master
 + b186ac8...d967634 refs/pull/250/merge -> refs/pull/250/merge  (forced update)
 + 41f0990...fc8af91 refs/pull/251/merge -> refs/pull/251/merge  (forced update)
 + 44c5426...fc4f423 refs/pull/261/merge -> refs/pull/261/merge  (forced update)
 + 36a4973...8d269bc refs/pull/316/merge -> refs/pull/316/merge  (forced update)
 + 9fe28fe...bf4a666 refs/pull/317/merge -> refs/pull/317/merge  (forced update)
 + 7e29bc2...d48ed27 refs/pull/323/merge -> refs/pull/323/merge  (forced update)
 + c908963...959881d refs/pull/339/merge -> refs/pull/339/merge  (forced update)
 + 23c19f4...a0d6bf2 refs/pull/349/merge -> refs/pull/349/merge  (forced update)
 + 951f7d4...acb17f4 refs/pull/364/merge -> refs/pull/364/merge  (forced update)
 + 2768b40...cb3adc9 refs/pull/369/merge -> refs/pull/369/merge  (forced update)
 + cbf7215...a8f708e refs/pull/372/merge -> refs/pull/372/merge  (forced update)
 + 2148e5e...a36c965 refs/pull/374/merge -> refs/pull/374/merge  (forced update)
 + 69e6113...dc7d7df refs/pull/383/merge -> refs/pull/383/merge  (forced update)
 + 4edf1d2...7f89945 refs/pull/386/merge -> refs/pull/386/merge  (forced update)
 + 2fe332c...9a8ff62 refs/pull/387/merge -> refs/pull/387/merge  (forced update)
 + b68944c...3a81981 refs/pull/389/merge -> refs/pull/389/merge  (forced update)
 + 433b8ba...074dabf refs/pull/392/merge -> refs/pull/392/merge  (forced update)
 + 31053ce...32415d5 refs/pull/399/merge -> refs/pull/399/merge  (forced update)
 + 0594d0e...2d1ae9b refs/pull/408/merge -> refs/pull/408/merge  (forced update)
 + e94e93a...2bc3b3b refs/pull/425/merge -> refs/pull/425/merge  (forced update)
 + 763be5c...673b74b refs/pull/428/merge -> refs/pull/428/merge  (forced update)
 + 0f0b68b...4c0714a refs/pull/429/merge -> refs/pull/429/merge  (forced update)
 + 6775051...5d396dd refs/pull/430/merge -> refs/pull/430/merge  (forced update)
 + c8bf518...a68515e refs/pull/431/merge -> refs/pull/431/merge  (forced update)
  -> Found kicad-doc-HEAD.tar.gz
==> Validating source files with md5sums...
    kicad ... Skipped
    kicad-i18n ... Skipped
    kicad-libs ... Skipped
    kicad-doc-HEAD.tar.gz ... Skipped
==> Extracting sources...
  -> Creating working copy of kicad-source-mirror git repo...
Reset branch 'makepkg'
  -> Creating working copy of kicad-i18n git repo...
From /home/Mike/MINGW-packages/mingw-w64-kicad-git/kicad-i18n
   62fed86..59adf27  master     -> origin/master
Reset branch 'makepkg'
  -> Creating working copy of kicad-library git repo...
From /home/Mike/MINGW-packages/mingw-w64-kicad-git/kicad-libs
   e571896..02bca80  master     -> origin/master
Reset branch 'makepkg'
  -> Extracting kicad-doc-HEAD.tar.gz with bsdtar
==> Starting pkgver()...
==> Updated version: mingw-w64-x86_64-kicad-git r6608.ba038ac-1
==> ERROR: A package has already been built. (use -f to overwrite)

The last time I tried this was about a year ago and it worked flawlessly.

Mike

mikewilt, 2016-03-05:
And another bit of output- the tail end of the output from the initial run:

[ 38%] ←[32mBuilding CXX object common/CMakeFiles/common.dir/geometry/shape_collisions.cpp.obj←[0m
[ 38%] ←[32mBuilding CXX object common/CMakeFiles/common.dir/geometry/shape_file_io.cpp.obj←[0m
[ 38%] ←[32m←[1mLinking CXX static library libcommon.a←[0m
[ 38%] Built target common
make: *** [all] Error 2
Makefile:127: recipe for target 'all' failed
==> ERROR: A failure occurred in build().
    Aborting...

Mike

nickoe, 2016-03-05:
I can't see tge actual build eeror, please try a clean build.

mikewilt, 2016-03-06:
I copied out the error I saw with a little more context above:

C:/Users/Mike/Documents/kicad-winbuilder-master/msys64/mingw32/bin/ranlib.exe: libpolygon.a: Permission denied
[ 11%] ←[32mBuilding CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/flag.cpp.obj←[0m
make[2]: *** [polygon/libpolygon.a] Error 1
polygon/CMakeFiles/polygon.dir/build.make:172: recipe for target 'polygon/libpolygon.a' failed
make[2]: *** Deleting file 'polygon/libpolygon.a'
CMakeFiles/Makefile2:1767: recipe for target 'polygon/CMakeFiles/polygon.dir/all' failed
make[1]: *** [polygon/CMakeFiles/polygon.dir/all] Error 2

There are tens of thousands of lines of output that look happy.  If there is more specific information in some log files please tell me where to look.

Mike

nickoe, 2016-03-06:
Yes, that looks more like the error that causes the build to fail. I don't know why it says permissin denied for libpolygon.a.  All I can suggest right now is it make a clean build.

nickoe, 2016-09-18:
Closing this issue because there have been no activity for a long time.

closed 2016-09-18 by nickoe