KiCad/kicad-winbuilder #35: Build Fails on windows 10

hyperbyte, 2016-03-15:
Hello, everyone

I'm trying to build kicad and i'm getting some tool configuration errors 
this is a clean instalation of windows 10 and MySys2 is up to date

Output
D:\GitHub\KiCad-Winbuilder>make_all.bat
-- HOME_DIR D:/GitHub/KiCad-Winbuilder/msys64/home/pmgomes
-- KICAD_PACKAGE_SOURCE_DIR D:/GitHub/KiCad-Winbuilder/msys64/home/pmgomes/MINGW-packages/mingw-w64-kicad-git/
-- MSYS2 user name is: $USERNAME=pmgomes
-- Running cd "D:/GitHub/KiCad-Winbuilder/msys64/home/pmgomes/MINGW-packages/mingw-w64-kicad-git" &&  makepkg-mingw -s --noconfirm
==> Making package: mingw-w64-x86_64-kicad-git r6494.edc47c3-1 (Tue, Mar 15, 2016  2:23:53 PM)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating kicad git repo...
Fetching origin
  -> Updating kicad-i18n git repo...
Fetching origin
  -> Updating kicad-libs git repo...
Fetching origin
  -> Found kicad-doc-HEAD.tar.gz
==> Validating source files with md5sums...
    kicad ... Skipped
    kicad-i18n ... Skipped
    kicad-libs ... Skipped
    kicad-doc-HEAD.tar.gz ... Skipped
==> Extracting sources...
  -> Creating working copy of kicad-source-mirror git repo...
Reset branch 'makepkg'
  -> Creating working copy of kicad-i18n git repo...
Reset branch 'makepkg'
  -> Creating working copy of kicad-library git repo...
Reset branch 'makepkg'
  -> Extracting kicad-doc-HEAD.tar.gz with bsdtar
==> Starting pkgver()...
==> Updated version: mingw-w64-x86_64-kicad-git r6622.b116318-1
==> Removing existing $pkgdir/ directory...
==> Starting build()...
CMake Error: CMake was unable to find a build program corresponding to "MSYS Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.-- Configuring incomplete, errors occurred!

CMake Error: CMake was unable to find a build program corresponding to "MSYS Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_AR was not found, please set to archive program.
==> ERROR: A failure occurred in build().
    Aborting...

D:\GitHub\KiCad-Winbuilder>

nickoe, 2016-03-15:
Hmm, that looks strange. Maybe the toolchain packages are not installed? Could you try to open the mingw shell from within the msys2 directry and then do `pacman -Ss mingw-w64-x86_64-toolchain` to see if that package is installed?

hyperbyte, 2016-03-30:
sorry for the late reply.
I think all the tools are installed but here is the output from the pacman command

Thanks
Pedro Gomes

```
$ pacman -Ss mingw-w64-x86_64-toolchain
mingw64/mingw-w64-x86_64-binutils 2.26-21 (mingw-w64-x86_64-toolchain) [installed]
    A set of programs to assemble and manipulate binary and object files (mingw-w64)
mingw64/mingw-w64-x86_64-crt-git 5.0.0.4627.03684c4-1 (mingw-w64-x86_64-toolchain) [installed]
    MinGW-w64 CRT for Windows
mingw64/mingw-w64-x86_64-gcc 5.3.0-2 (mingw-w64-x86_64-toolchain) [installed]
    GNU Compiler Collection (C,C++,OpenMP) for MinGW-w64
mingw64/mingw-w64-x86_64-gcc-ada 5.3.0-2 (mingw-w64-x86_64-toolchain) [installed]
    GNU Compiler Collection (Ada) for MinGW-w64
mingw64/mingw-w64-x86_64-gcc-fortran 5.3.0-2 (mingw-w64-x86_64-toolchain) [installed]
    GNU Compiler Collection (Fortran) for MinGW-w64
mingw64/mingw-w64-x86_64-gcc-libgfortran 5.3.0-2 (mingw-w64-x86_64-toolchain) [installed]
    GNU Compiler Collection (libgfortran) for MinGW-w64
mingw64/mingw-w64-x86_64-gcc-libs 5.3.0-2 (mingw-w64-x86_64-toolchain) [installed]
    GNU Compiler Collection (libraries) for MinGW-w64
mingw64/mingw-w64-x86_64-gcc-objc 5.3.0-2 (mingw-w64-x86_64-toolchain) [installed]
    GNU Compiler Collection (ObjC,Obj-C++) for MinGW-w64
mingw64/mingw-w64-x86_64-gdb 7.11-1 (mingw-w64-x86_64-toolchain) [installed]
    GNU Debugger (mingw-w64)
mingw64/mingw-w64-x86_64-headers-git 5.0.0.4627.53be55d-1 (mingw-w64-x86_64-toolchain) [installed]
    MinGW-w64 headers for Windows
mingw64/mingw-w64-x86_64-libmangle-git 5.0.0.4509.2e5a9a2-1 (mingw-w64-x86_64-toolchain) [installed]
    MinGW-w64 libmangle
mingw64/mingw-w64-x86_64-libwinpthread-git 5.0.0.4573.628fdbf-1 (mingw-w64-x86_64-toolchain) [installed]
    MinGW-w64 winpthreads library
mingw64/mingw-w64-x86_64-make 4.1.2351.a80a8b8-1 (mingw-w64-x86_64-toolchain) [installed]
    GNU make utility to maintain groups of programs (mingw-w64)
mingw64/mingw-w64-x86_64-pkg-config 0.29.1-1 (mingw-w64-x86_64-toolchain) [installed]
    A system for managing library compile/link flags (mingw-w64)
mingw64/mingw-w64-x86_64-tools-git 5.0.0.4592.90b8472-1 (mingw-w64-x86_64-toolchain) [installed]
    MinGW-w64 tools
mingw64/mingw-w64-x86_64-winpthreads-git 5.0.0.4573.628fdbf-1 (mingw-w64-x86_64-toolchain) [installed]
    MinGW-w64 winpthreads library
```

joernesdohr, 2016-06-21:
@hyperbyte did you manage to solve this issue? I'm having the same issue and so far am unsuccessful in building KiCad myself.

nickoe, 2017-12-14:
Issue too silent for too long, works fine for me on windows 10. Closing. Feel free to open a new issue with a more recent version of kicad-winbuilder.

closed 2017-12-14 by nickoe