KiCad/kicad-winbuilder #43: Missing dll(s) prevent KiCad from starting

milzec, 2016-09-06:
After fresh build (Win7 Pro, 64bit) and installing, KiCad won't start.
The file kicad.exe reports an error:
"The program can't start because libnghttp2-12.dll is missing from your computer. Try reinstalling the program to fix this problem".

I am using the latest Winbuilder (commit 488af83).

nickoe, 2016-09-06:
So did you enable KICAD_SPICE by any chance?

milzec, 2016-09-09:
Hi nickoe,

Silly question: how to check if KICAD_SPICE is enabled?
Feel free to reject this issue due to my incomplete actions.

BR,
Miloje

nickoe, 2016-09-09:
Ignore the comment about spice. I just thought it was ng-spice because of the ng. 

It seems that there is a package called mingw-w64-x86_64-nghttp2.

Please try to start the msys shell and check if it is installed. Åleade paste the output of `pacman -S nghttp2`

milzec, 2016-09-09:
Hi nickoe,

Here is the output:
$ pacman -Ss nghttp2
mingw32/mingw-w64-i686-nghttp2 1.13.0-1 [installed]
    Framing layer of HTTP/2 is implemented as a reusable C library (mingw-w64)
mingw64/mingw-w64-x86_64-nghttp2 1.13.0-1 [installed]
    Framing layer of HTTP/2 is implemented as a reusable C library (mingw-w64)

I've deleted everything so I'll run the latest commit of winbuilder once again.

BR

nickoe, 2016-09-09:
Hmm, sonehow it looks for nghttp2 1.14, while you have 1.13 installed. Looks like there have been performed a partial upgrade.

You could likely just run pacman -Sy   and then pacman -S <nghttp2pkgnamehere>    and then rebuild.

milzec, 2016-09-12:
Hi nickoe,

It doesn't resolve this. It says that minigw32, minigw64 and msys are up to date but still remains with 1.13 installed. I'll try to look for the problem.

nickoe, 2016-09-12:
Ok, I can see that the dll is indeed provided by those versions of nghttp2 you have installed, but I don't understand what it is used for. I would like to understand that.

I mean, I don't copy that dll on the nightlies and they work without it.

nickoe, 2017-01-14:
Apparently it seems to be curl that requires it...

closed 2017-01-14 by nickoe

nickoe, 2017-01-14:
https://github.com/Alexpux/MINGW-packages/commit/72a7e23e0cb5a6d55e75ccabd652c8215593b6c2#diff-b1c5d370af281c25aed48dd38c3dda51