KiCad/kicad-winbuilder #44: During build OCE not downloaded

gusber, 2016-09-13:
running make_all.bat gave an error that OCE is not downloaded as part of the winbuilder.

After downloading the packages and build-checks are in progress I get this error output:
CMake Error at CMakeLists.txt:548 (find_package):
  By not providing "FindOCE.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OCE", but
  CMake did not find one.

  Could not find a package configuration file provided by "OCE" (requested
  version 0.16) with any of the following names:

    OCEConfig.cmake
    oce-config.cmake

  Add the installation prefix of "OCE" to CMAKE_PREFIX_PATH or set "OCE_DIR"
  to a directory containing one of the above files.  If "OCE" provides a
  separate development package or SDK, be sure it has been installed.

nickoe, 2016-09-13:
Did you install the oce packages with pacman manually?

gusber, 2016-09-13:
no, just cloned the kicad-winbuilder and ran make_all.bat

nickoe, 2016-09-13:
So please try to pull and rerun. I added oce as a depends on the PKGBUILD, so it should install that now.

gusber, 2016-09-13:
Just a slight typo in PKGBUILD line 20:
         "${MINGW_PACKAGE_PREFIX}-libxslt")
         "${MINGW_PACKAGE_PREFIX}-oce")

Should be:

         "${MINGW_PACKAGE_PREFIX}-libxslt"
         "${MINGW_PACKAGE_PREFIX}-oce")

nickoe, 2016-09-13:
Ahh yeah, that was stupid of me. I have pushed a fixed version.

gusber, 2016-09-13:
It builds now, thank you. will try with a fresh clone tomorrow.

gusber, 2016-09-14:
The initial build works. 
This problem might be a kicad problem though but,
during the build the MINGW-packages\mingw-w64-kicad-git\src\build-x86_64-w64-mingw32\plugins\3d\oce\CMakeFiles\s3d_plugin_oce.dir\build.make
refers to C:/building/msys64/mingw64/lib/libfreetype.dll.a

I added the oce-win-bundle libraries to the kicad winbuilder script
`
         "${MINGW_PACKAGE_PREFIX}-libxslt"
         "${MINGW_PACKAGE_PREFIX}-freetype"
         "${MINGW_PACKAGE_PREFIX}-freeimage"
         "${MINGW_PACKAGE_PREFIX}-gl2ps"
         "${MINGW_PACKAGE_PREFIX}-oce")`

and then modified the MINGW-packages\mingw-w64-kicad-git\src\build-x86_64-w64-mingw32\plugins\3d\oce\CMakeFiles\s3d_plugin_oce.dir\build.make to point to the correct location of libfreetype.dll.a

Then kicad builds OK.

nickoe, 2016-09-14:
What is the "oce-win-bundle libraries"?

What is the "correct" location of libfreetype.dll.a?

gusber, 2016-09-14:
if you build oce under windows you should also build oce-win-bundle which is just gl2ps, freetype and freeimage. 

the 'correct' location i pointed the makefile to find libfreetype.dll.a is kicad-winbuilder\msys64\mingw64\lib

nickoe, 2016-09-14:
No, gl2ps, freetype and freeimage should be provided by msys2.

I should not be nescesary to change the path in the build.make.

gusber, 2016-09-14:
ok, this is odd. is there any easy way to enable debugging messages using the kicad-winbuilder?
(not build debug version of kicad :). i have tried but failed with and without "${MINGW_PACKAGE_PREFIX}-freetype" "${MINGW_PACKAGE_PREFIX}-freeimage" "${MINGW_PACKAGE_PREFIX}-gl2ps" )

nickoe, 2016-09-14:
Those packages are supposed to be installed with oce if they are a runtime dependency.

Can you check if they are actually installed in the msys2 environment inside the kicad-winbuilder directory? Really only freetype is a dependency because I have not built OCE with gl2ps or freeimage.

So you should be able to check that it is installed with:
`pacman -Qs freeimage`

And by your earlier comments it does indeed sugges that it is installed, because you say that it is in `kicad-winbuilder\msys64\mingw64\lib`. What I don't understand is that it originally reffered to `C:/building/msys64/mingw64/lib/libfreetype.dll.a`. Where does that "building" dir come from?

There should be log files in the .log folder.

gusber, 2016-09-14:
I have no idea where building comes from. This is why i would like to have a debug output from the script. 
The logs in the .log folder only have stdout not stderr.

i get to `[ 45%] Built target common` then 
`make: *** [Makefile:128: all] Error 2`
`==> ERROR: A failure occurred in build().`
`    Aborting...`

`pacman -Qs freeimage` return 1
libfreetype.dll.a is installed. Will continue tomorrow.

nickoe, 2016-09-14:
I think the logs also have stderr, but in tat case I think the error is further up in the log.

gusber, 2016-09-14:
a text search through all logs on 'error' and 'warning' only gives me information about p5-error packages and compiling warning.cpp. and from pacman update to say that msys should be started with the -mingw64 flag. 
`WARNING: the shell starting scripts have been unified.`
...
`MSYS2_ROOT\msys2_shell.cmd -mingw64`

gusber, 2016-09-15:
I can't figure out why CMake cannot find the correct path to freetype.dll.a while building kicad's oce plugin. This seems to be the problem though. As i mentioned before, running make_x86_64.bat i get to 45% and then `make: *** [Makefile:128: all] Error 2` without no further clues. 

launching msys and running the PKGBUILD script manually step by step i found out that the plugins/3d/oce/CMakeFiles/s3d_plugin_oce.dir/build.make have :/building/msys64/mingw64/lib/libfreetype.dll.a as a dependancy. 

Editing build.make to point to libfreetype.dll.a to the kicad-winbuilder mingw bin directory fixes the build. (i.e. kicad is built with make and then i can run the make DESTDIR=.. install )

Finally, trying to launch pcbnew there's an error with not being able to read the _pcbnew.kiface library. After playing around with the dll-files in the mingw/bin directory i found that libpcre dlls were necessary. 

Now the _pcbnew.kiface is loaded ok (I think) and a new error occurs:

> Mismatch between the program and library build versions detected. The library used 3.0 (wchar_t, compiler with C++ ABI 1010, wx containers, compatible with 2.8), and wxPython used 3.0 (wchar_t, compiler C++ ABI 1009, wx containers, compatible with 2.8)

So i supposed wxPython need to be rebuilt too.

nickoe, 2016-09-16:
Could you try to share the complete build log?

When you install anywhere thant to not a standard location you need to make sure the exes can fild the kifcace, for example by setting the LD_LIBRARY_PATH to point at them. Or as we do in the installer copy everything to the same dir as the exe's.

I have not yet found the wx ABI mismatch to cause any problems with KiCad yet, it is just a warning after all.

gusber, 2016-09-16:
I will share all the logs. In progress with a fresh clone now. 

regarding the ABI problem the pop-up says Fatal Error, not warning and then kicad crashes.
[https://ibin.co/2vGlQsZmTtcI.jpg](url) [https://ibin.co/2vGlkrSPpBpR.jpg](url)

will share logs in a bit..

gusber, 2016-09-16:
cmd.exe output log: [cmd.exe log](http://pastebin.com/4DANAnvL)
makepkg log: [makepkg](http://pastebin.com/SB3LBuQM)
pacman_bash log: [pacman_bash](http://pastebin.com/XZfwhmqx)
pacman_bash2 log:  there is nothing to do
pacman_initial log: [pacman_initial](http://pastebin.com/YSVyTueh)'
pacman_required_packages log: [pacman_required_packages](http://pastebin.com/h60k4312)
pacman_required_packages_update log: [pacman_required_packages_update](http://pastebin.com/aWi7nhet)
pacman_update log: [pacman_update](http://pastebin.com/sBRFPGWA)

I cannot verify that this error seen in cmd.exe output is due to the fact that Cmake finds the wrong path to libfreetype but i would suspect so.

gusber, 2016-09-16:
I did `make DESTDIR=/home/gusber/out install` and then ran the copydll.sh but just the copystuff() part. 
All dlls were copied but libnghttp and the libpcre

Having had a look with a dependency viewer i found that
libgraphite2.dll
libicudt57.dll
libicuin57.dll
libicuio57.dll
libicule57.dll
libiculx57.dll
libicutest57.dll
libicutu57.dll
libicuuc57.dll
libpcre-1.dll
libpcre16-0.dll
libpcre32-0.dll
libpcrecpp-0.dll
libpcreposix-0.dll

were missing from the out/bin/ directory

nickoe, 2016-09-17:
Ok, I guess this is the reason that the ABI check is not fatal for me.

https://git.archlinux.org/svntogit/packages.git/tree/trunk/make-abicheck-non-fatal.patch?h=packages/wxgtk

nickoe, 2018-07-08:
This issue has been silent for a long time, and it works with latest version of kicad-winbuilder.

closed 2018-07-08 by nickoe