KiCad/kicad-winbuilder #45: Cannot build Kicad - no such file or directory

trayres, 2016-12-03:
C:\Users\tayres\Downloads\kicad-winbuilder-master>make_all.bat
-- HOME_DIR
-- KICAD_PACKAGE_SOURCE_DIR /MINGW-packages/mingw-w64-kicad-git/
-- MSYS2 user name is: $USERNAME=
-- Running cd "/MINGW-packages/mingw-w64-kicad-git" &&  makepkg-mingw -s --nocon
firm
2>&1: line 0: cd: /MINGW-packages/mingw-w64-kicad-git: No such file or directory

Went into msys64 folder, ran msys2 shell, then installed the tool groups for x86 and x86_64 manually. These failed multiple times for various network reasons, but finally installed. 

But I'm still getting the error.

nickoe, 2016-12-03:
Does your windows username contain spaces or non ascii characters?

trayres, 2016-12-03:
The windows username is tayres. I checked the tickets but I didn't see a fitting issue already opened.

trayres, 2016-12-05:
/MINGW-packages/ is not getting created - does that help narrow it down?

nickoe, 2016-12-05:
I think the problem is that HOME_DIR is not set. Do you see any folder in the msys64/home?

trayres, 2016-12-05:
No I don't; I have an MSYS2 installation that isn't the one provided by Kicad-Winbuilder, is it possible they're interfering?

I could set HOME_DIR with the batch file that kicks off the installation in make_i686, but I'm not sure if that's the best way to fix it with CMAKE driving the show.

trayres, 2016-12-08:
I tried to export HOME_DIR in the batch files that kick things off, but that didn't work.

nickoe, 2016-12-08:
Try to hardcode it in the cmake file:
`file( GLOB HOME_DIR "${CMAKE_SOURCE_DIR}/${MSYS2}/home/*" )`
replace with
`file( GLOB HOME_DIR "${CMAKE_SOURCE_DIR}/msys64/home/tayres" )`

trayres, 2016-12-08:
Result:

```
C:\Users\tayres\Downloads\kicad-winbuilder-master>make_all.bat
-- HOME_DIR
-- KICAD_PACKAGE_SOURCE_DIR /MINGW-packages/mingw-w64-kicad-git/
-- MSYS2 user name is: $USERNAME=
-- Running cd "/MINGW-packages/mingw-w64-kicad-git" &&  makepkg-mingw -s --nocon
firm
2>&1: line 0: cd: /MINGW-packages/mingw-w64-kicad-git: No such file or directory
```

nickoe, 2016-12-08:
Mmm, maybe try set instead of file:
`set( HOME_DIR "${CMAKE_SOURCE_DIR}/msys64/home/tayres" )`

trayres, 2016-12-09:
That did the trick!
Now I'm getting stuck here; attached are CMakeErrorLog and CMakeOutput

`-- Found wxWidgets: -LC:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/m
ingw64/lib;;;-pipe;-Wl,--subsystem,windows;-mwindows;-lwx_mswu_gl-3.0;-lwx_mswu_
aui-3.0;-lwx_mswu_adv-3.0;-lwx_mswu_html-3.0;-lwx_mswu_core-3.0;-lwx_baseu_net-3
.0;-lwx_baseu-3.0;-lwx_baseu_xml-3.0;-lwx_mswu_stc-3.0 (found suitable version "
3.0.2", minimum required is "3.0.0")
-- Found OpenGL: opengl32
-- Found Glew: C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/mingw64/
lib/libglew32.dll.a
-- Check for installed GLEW -- found
-- Found GLM: C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/mingw64/i
nclude (found suitable version "0.9.8.3", minimum required is "0.9.5.1")
-- Found CURL: C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/mingw64/
lib/libcurl.dll.a (found version "7.51.0")
-- Found PkgConfig: C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/min
gw64/bin/pkg-config.exe (found version "0.29.1")
-- Checking for module 'cairo'
--   Found cairo, version 1.15.2
-- Found Cairo: C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/mingw64
/lib/libcairo.dll.a (found suitable version "1.15.2", minimum required is "1.8.8
")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   context
--   system
--   thread
--   chrono
--   date_time
--   atomic

*** NGSPICE library missing ***
Most of ngspice packages do not provide the required libngspice library.
You can either compile ngspice configured with --with-ngshared parameter
or run a script that does the job for you:
  wget https://orson.net.pl/pub/libngspice/get_libngspice_so.sh
  chmod +x get_libngspice_so.sh
  ./get_libngspice_so.sh
  sudo ./get_libngspice_so.sh install

CMake Error at C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/mingw64/
share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find ngspice (missing: NGSPICE_INCLUDE_DIR NGSPICE_LIBRARY)
Call Stack (most recent call first):
  C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/mingw64/share/cmake-3
.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  CMakeModules/Findngspice.cmake:30 (find_package_handle_standard_args)
  CMakeLists.txt:553 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/home/tayres/M
INGW-packages/mingw-w64-kicad-git/src/build-x86_64-w64-mingw32/CMakeFiles/CMakeO
utput.log".
See also "C:/Users/tayres/Downloads/kicad-winbuilder-master/msys64/home/tayres/M
INGW-packages/mingw-w64-kicad-git/src/build-x86_64-w64-mingw32/CMakeFiles/CMakeE
rror.log".
==> ERROR: A failure occurred in build().
    Aborting...`

[CMakeError.log.txt](https://github.com/KiCad/kicad-winbuilder/files/642849/CMakeError.log.txt)
[CMakeOutput.log.txt](https://github.com/KiCad/kicad-winbuilder/files/642850/CMakeOutput.log.txt)

Is this a new issue or a continuation of this one?

nickoe, 2016-12-09:
Ahh, ok, that is expected. Tou can either install oce or disable that in tge pkgbuild.

trayres, 2016-12-09:
I opened the kicad-winbuilder msys2_shell, and I do see 
`mingw64/mingw-w64-x86_64-oce 0.17.2-4 [installed]`

nickoe, 2016-12-10:
I am not sure why I asked about oce... the problem is ngspice. Which is built by the PKGBUILD-ngspice-git. But you could alos just disbale spice in the normal PKGBUILD if you like.

trayres, 2016-12-15:
I'm sorry this is a stupid question - how do I launch PKGBUILD-ngspice-git so I get the DLL'd ngspice to run with KICAD?

nickoe, 2018-04-01:
@trayres Still active?

closed 2018-04-01 by nickoe