KiCad/kicad-winbuilder #46: Windows Installer: text about Wing3D in the infobox after finishing the installation isn't complete readable

tijuca, 2016-12-10:
After a long time I installed the recent nightly amd64 version of the Windows version for KiCad. All went fine. Except the textbox about the indormation for the Wing3D package is to small or the text is to long. :-) Preferably the textbox needs to be more flexable as the available space in the box isn't used complete.

[image]

nickoe, 2016-12-10:
I think I would actually rather remove it. I don't think any one uses wings3d anymore and we suport iges and step models in the master branch now.

easyw, 2016-12-10:
May be it would be useful to suggest FreeCAD both for VRML and STEP models... proven to be most compatible sw (same 3D mechanical kernel OCE), already used by many 3D developers, and open source too

Maurice

tijuca, 2016-12-11:
Some kind of information about additional tools or software at this step in the installation is at least a good thing. But as people mostly an a "rush" at this point and don't get really noticed I would also put such a information in the info box (Help -> About KiCad) in the KiCad project manager. This must not be much or explicitly detailed, this could be done on the website of KiCad. There is really enough space to explain the various tools and technicals. Also there its probably easier to maintain downloads links and howtos. I still miss a easy guide to create basic 3D models and haven't found something actual

nickoe, 2016-12-11:
What about simply removing it, does users ever click on stuff like that post install? It is only there because it was there before I adopted the installer.

But we might wat to add FreeCAD on http://kicad-pcb.org/external-tools/

tijuca, 2016-12-11:
Then I would add some small information about additional external tools that can be found on that URL.

closed 2016-12-15 by nickoe