KiCad/kicad-winbuilder #52: Can't build, failure after libbitmaps.a

chalecampb, 2017-05-09:
Hi,

Tried several times to build. Previous times, I got to this same point. This time, I applied both open patches and it worked out of the box (didn't need to remove some lines referring to /last_error or hardcode the HOME_DIR path). I had attributed the below failure to having messed with the .cmake file, but the below error seems to happen regardless. 

[ 36%] Linking CXX static library libbitmaps.a
[ 36%] Built target bitmaps
make: *** [Makefile:128: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
CMake Error at KiCad-Winbuilder.cmake:289 (message):
  Error running cd
  "F:/Source/kicad-winbuilder/kicad-winbuilder/msys64/home/Campbell/MINGW-packages/mingw-w64-kicad-git"
  && makepkg-mingw -s --noconfirm

   Output in: F:/Source/kicad-winbuilder/kicad-winbuilder/.logs/last_error
Call Stack (most recent call first):
  KiCad-Winbuilder.cmake:357 (execute_msys2_bash)

Any assistance would be appreciated.

Thanks,
-Alex

chalecampb, 2017-05-09:
Also as a side note, I can't help but to feel that if there is an error, there should be some way to see it. If it's a missing reference or something I can probably get around it, but this just says "build failed", not much else I can do with the information.

kelogs, 2017-08-12:
Getting a similar issue here.

```
[ 38%] Building CXX object plugins/3d/vrml/CMakeFiles/s3d_plugin_vrml.dir/x3d/x3
d_shape.cpp.obj
[ 38%] Building CXX object plugins/3d/vrml/CMakeFiles/s3d_plugin_vrml.dir/x3d/x3
d_transform.cpp.obj
[ 38%] Linking CXX shared module libs3d_plugin_vrml.dll
[ 38%] Built target s3d_plugin_vrml
make: *** [Makefile:130: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
CMake Error at KiCad-Winbuilder.cmake:286 (message):
  Error running cd
  "C:/kicad-winbuilder-master/msys32/home/kellogs/MINGW-packages/mingw-w64-kicad
-git"
  && export CARCH=i686 && TERM=vt220 makepkg-mingw -s --noconfirm

   Output in: C:/kicad-winbuilder-master/.logs/last_error
Call Stack (most recent call first):
  KiCad-Winbuilder.cmake:356 (execute_msys2_bash)

```

With some relevant lines in last_error:

```
[ 26%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/redo.cpp.obj
[ 26%] Building CXX object utils/kicad2step/CMakeFiles/kicad2step.dir/__/__/common/streamwrapper.cpp.obj
[ 26%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/red.cpp.obj
make[2]: *** No rule to make target 'C:/building/msys64/mingw32/lib/libfreetype.dll.a', needed by 'utils/kicad2step/kicad2step.exe'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 26%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/reload2.cpp.obj
[ 26%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/reload.cpp.obj
[ 26%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/render_mode.cpp.obj
[ 27%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/reset_text.cpp.obj
[ 27%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/resize_sheet.cpp.obj
[ 27%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/rescue.cpp.obj
make[1]: *** [CMakeFiles/Makefile2:2945: utils/kicad2step/CMakeFiles/kicad2step.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 27%] Building CXX object common/CMakeFiles/gal.dir/gl_context_mgr.cpp.obj
[ 27%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/right.cpp.obj
[ 27%] Building CXX object bitmaps_png/CMakeFiles/bitmaps.dir/cpp_26/rotate_ccw.cpp.obj
```

Building on Win 7 32 bit; a previous build also stopped at 38% but the errors were a bit different.

Thank you

nickoe, 2018-04-01:
I can't reproduce, are you still able to reproduce this?

closed 2018-04-01 by nickoe