KiCad/kicad-winbuilder #53: Windows Nightly needs splitting into application and data installers

davidsrsb, 2017-07-19:
Windows Nightly builds have jumped to 670MB since 16th July.
I suggest that the time has come for the Windows builds to be split into two components, the KiCad application and the data libraries. New users would install both. Existing nightly users can skip daily updating of the bigger library build until a necessary change comes along

davidsrsb, 2017-07-20:
That would work. Then most Nightly experimenters could just fetch the slim version. Presumably this would be MUCH smaller.
There are Nightly users who have frozen at a known stable application build, but they are the kind that make their own libraries.

nickoe, 2017-07-20:
I have sort of done this for the patch builds, but the installer still has the menu option to instsll rhe parts not contained in the installer, which is not ideal. An option should be made to the scripts to do a slim or lite version. I don't know what word would be best to describe it.

JoanTheSpark, 2017-09-07:
What is the intent of the nightlies? - For users to test the program, no?
Anyone who installs them will be able to fetch the libraries via a stable install or manual download.
Anyone who can't do that shouldn't be running nightlies at the moment.
IMHO.

poeschlr, 2017-10-17:
As far as i know ubuntu already separates the package for kicad (the program) and the one for the lib. It might be an option to talk to the guy who did this. Maybe he has an idea how do to it for windows as well.

nickoe, 2017-10-17:
It is not hard to do, it just needs to be done properly.

easyw, 2017-10-18:
It would also be nice if the suggestion to install Wings3D would be removed from the Win installer or changed to suggest to install FreeCAD.. 
I think this is causing confusion for users that want to create some 3D models

davidsrsb, 2017-10-19:
Better both. Wings3D models are better than STEP when you want textures

easyw, 2017-10-19:
Texture is not available in Kicad. If you talk about material properties, then FreeCAD can handle those correctly. In fact all the new 3d library is based on FreeCAD and none of them in Wings3d.

easyw, 2017-10-19:
Texture is not available in Kicad. If you talk about material properties, then FreeCAD can handle those correctly. In fact all the new 3d library is based on FreeCAD and none of them on Wings3d.

davidsrsb, 2017-10-20:
4.0.x Stable does not support textures.
Nightly does have support for shininess etc, for some time.
See this old link for a simple test project
https://forum.kicad.info/t/does-the-3d-viewer-support-textures/1577/19
FreeCAD can generate these too, so Wings3d is going obsolete

easyw, 2017-10-20:
Texture is different from material properties like shininess etc. which are supported both in Kicad 4 and in nightly.
Those material properties are rendered in a improved way in nightly...
Also for being able to generate step and wrl with material properties, FC is the right sw to model 3d parts.
Wings3d is for sure obsolete :)

bombledmonk, 2018-08-02:
Just following up linking the related exchange on the mailing list.  
Summary:  The current 5.0 installer is 1GB, while the Kicad 4.07 installer was something like 107MB.  This is mostly due to the 3d models being included.  It would be nice, to split the installer to ease download requirements for end users.  
https://lists.launchpad.net/kicad-developers/msg36982.html

(Selfishly, this would also solve our silly restricted corporate environment install issues on 100+ workstations)

FabianInostroza, 2018-08-03:
Maybe also would be nice to be able to choose to install only step models, wrl models or both.

easyw, 2018-08-04:
@FabianInostroza 
installing only STEP models would make the 3d viewer fail to display any of the kicad standard library model.
The footprints are built to point to WRL file format to  give the 3d viewer the best result.
Moreover installing only WRL will make the STEP exporter to fail the mechanical conversion from the kicad board to the mechanical assembly.

FabianInostroza, 2018-10-04:
@easyw mm, I though it tried changing the extension to .step if loading the wrl failed. Currently the 3d viewer can load step models, the automatic try of .step file is missing.

easyw, 2018-10-04:
@FabianInostroza 
you misunderstood the tip:
1) you must have both wrl and step models for your 3d part
2) insert the wrl model in kicad_pcb board footprints field
3) export kicad_pcb to STEP 
only the exporting process (point 3) is doing the substitution from wrl to step; but obviously this is working ONLY if both models are existing in your path.

FabianInostroza, 2018-10-04:
@easyw Well, If choose not to install step models is it reasonable for me to expect that the step export would not work, the same for wrl models.

But I see no reason to no include a step model in the export if the wrl is missing, that is something from the software side and cannot be addressed here.

Anyway, having a little installer (no libs) would be very nice, the only reason I use kicad in windows is to check if problems existing in linux are also in windows. Having to always download and install the full libraries is a little annoying for me (mostly because I dislike windows).

evanshultz, 2019-03-12:
Any progress on this? I completely agree that managing the libraries should be separate for KiCad nightly users and it's really a pain to download this huge binary of which only a small percentage (10%) is actually being used.

Another option would be to remove all libraries from the installer, and then if the user selects to install the libraries they can be downloaded during installation. I'm not advocating for this approach, just pointing out another method that some Windows software uses (including MS stuff, like, for example, a .NET framework).

@nickoe 
I understood from https://github.com/KiCad/kicad-winbuilder/issues/53#issuecomment-337167911 that you're unsure what is the best way to proceed. What information are you missing?

nickoe, 2019-03-12:
As per https://github.com/KiCad/kicad-winbuilder/issues/53#issuecomment-316653638 I want to update the nsi script to not have the install options page for nightlies and not have any other dependencies other than the translations.

What I need is time, or a proper patch and less time.

I am not sure what the best interface for this is, but I think an option to the copydlls.sh script to generate a lite version would be good. Then it should create two installers for each build it packages. The lite version should also package successfully if only a kicad build is provided.

Salamandar, 2019-08-19:
@evanshultz I don't like this approach. 

During an update, should I update everything by selecting the libraries too ? Or can I download them only sometimes ?

Also, I use the windows package manager Chocolatey, that bypasses the prompts with default values.

KiCAD really needs to provide 2 packages

* software + schematics + footprints
* 3d packages

or 3 packages
* software (340Mo on disk) and docs (280Mo on disk !!) ( = 720Mo)
* schematics (40Mo on disk) + footprint libraries (90Mo on disk)
* 3d packages (4.9Go on disk)

I'd go for the 2 packages version, as the schematics+modules libraries are not so huge.