KiCad/kicad-winbuilder #58: Just plain fails

BrianAtDocumentedDesigns, 2018-02-22:
I don't see many replies to the issues posted but I'll give it a try.

I unzipped into a user directory and got errors running pacman
-- Running pacman --noconfirm -Sy
mkdir: cannot change permissions of '/dev/shm': Permission denied
mkdir: cannot change permissions of '/dev/mqueue': Permission denied

A little while later I got this 
looking for conflicting packages...
:: msys2-runtime and catgets are in conflict. Remove catgets? [y/N] error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: msys2-runtime and catgets are in conflict
CMake Error at KiCad-Winbuilder.cmake:285 (message):
  Error running pacman --noconfirm --needed -S bash pacman pacman-mirrors
  msys2-runtime

   Output in: C:/Users/Brian_17/kicad-winbuilder-master/.logs/last_error
Call Stack (most recent call first):
  KiCad-Winbuilder.cmake:295 (execute_msys2_bash)

and the batchfile exited.

It seemed to go further then I got

upgrading pkgfile...
      0 [main] pacman 2060 fork: child -1 - forked process 13748 died unexpectedly, retry 0, exit code 0xC0000139, errno 11
      0 [main] pacman 2060 fork: child -1 - forked process 13748 died unexpectedly, retry 0, exit code 0xC0000139, errno 11
error: could not fork a new process (Resource temporarily unavailable)

Now, this is mentioned in the readme (for some reason kicad-builder is downloading an ancient version of MSYS2. 

The readme.txt file says

1. Launch msys2_shell.bat
2. run command 'update-core'
update-core is deprecated; its functionality is now handled by pacman; run pacman -Syuu
3. exit shell, now the rest of the Winbuilder process will work (although the pacman_initial log file should be removed to re-run the pacman updates)

But there is no msys2_shell.bat file.

There is an msys64 directory. Running msys2.exe in it and invoking update-core
gets you this 
$ update-core
-bash: update-core: command not found

Research on the web suggests 
update-core is deprecated; its functionality is now handled by pacman; run pacman -Syuu

Running pacman -Syuu has no effect (like typing enter).

Any effort to run make immediately fails

Here is a trace of my results

[Winbuilder trace.txt](https://github.com/KiCad/kicad-winbuilder/files/1748476/Winbuilder.trace.txt)

Thanks

nickoe, 2018-02-22:
I guess it is related to https://github.com/Alexpux/MSYS2-packages/issues/1141
Your post is hard to read, please consider to put snippets copied from logs or console in a codeblock to have proper formatting.

BrianAtDocumentedDesigns, 2018-02-22:
I'm not sure I understand about my post being hard to read. I did include the entire output from windows in the text file WIndbuilder Trace.txt - is that what you mean?

nickoe, 2018-02-22:
I updated your description, it is much more readable with some better text formatting.

nickoe, 2018-02-22:
The msys2 base package it downloads is the latest from msys2.org.

BrianAtDocumentedDesigns, 2018-02-22:
Thank you. I thought it was 2016 and Msys 2 had a very recent package.

I am not sure I understand the thread you linked to. Is there a resolution? I am not very advanced with respect to various linux tools: I wrote a utility for Kicad (renumKiCadPCB) and I am hoping to figure out how to integrate it into PCBNew.

Every time I try follow Windows build instructions (i.e. from Kicad or here) I hit a brick wall, even though I follow the instructions precisely.

Any help would be appreciated.

BrianAtDocumentedDesigns, 2018-02-23:
Is it possible the fact I have multiple Msys2 installations (associated with different Kicad builder solutions) is to blame?

nickoe, 2018-02-23:
I don't think so.

BrianAtDocumentedDesigns, 2018-02-23:
I do find it interesting that when I enter pacman -Syuu in Msys2 associated with kicad-winbuilder nothing happens 
BrianP@DESKTOP-FA1TEPK MSYS ~
$ pacman -Syuu

BrianP@DESKTOP-FA1TEPK MSYS ~

and yet when I do that under the Msys2 I installed associated with the Kicad make instructions (which fail) I get this 

BrianP@DESKTOP-FA1TEPK MSYS ~
$ pacman -Syuu
:: Synchronizing package databases...
 mingw32 is up to date
 mingw64 is up to date
 msys is up to date
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
 there is nothing to do

BrianP@DESKTOP-FA1TEPK MSYS ~

The behavior is different in response to the command and the path BrianP@DESKTOP-FA1TEPK MSYS is the same.

nickoe, 2018-03-26:
Hmm, it just worked for me today. Are you sure that your machine did not go to sleep while it ran? I had troble and I just left the machine when it worked, until I realized that windows got sleepy. I changed my power settings to never sleep and not I got at least one good build...

BrianAtDocumentedDesigns, 2018-03-27:
No, it did not go to sleep. I abandoned software and got the Kicad devs to investigate and update the directions for building under Msys2.