KiCad/kicad-winbuilder #60: Tkinter (part of python) does not work

baranovskiykonstantin, 2018-04-26:
I noticed that Tkinter (GUI module of python based on Tk) does not work on Windows. Part of it is present in _"path_to_kicad/lib/python2.7/lib-tk"_, but binary libraries tk and tcl is missing.
Attached patch includes tk and tcl dependencies to KiCad installation.
[0001-Tk-is-installed-with-KiCad-as-part-of-python.patch.txt](https://github.com/KiCad/kicad-winbuilder/files/1951626/0001-Tk-is-installed-with-KiCad-as-part-of-python.patch.txt)

nickoe, 2018-07-05:
Hello @baranovskiykonstantin 

I tested your patch, but it does not seem like it works as intended. Following some examples of tkinter hello worlds I already get an attribute error as follows:

```
from Tkinter import *
root = Tk()
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\KiCad\lib\python2.7\lib-tk/Tkinter.py", line 1814, in __init__
    baseName = os.path.basename(sys.argv[0])
AttributeError: 'module' object has no attribute 'argv'
```

Can you provide a minimal example you expect to work?

baranovskiykonstantin, 2018-07-07:
Hi @cpavlina koe.
I tested it outside of pcbnew, just execute python.exe from cmd and run next couple of commands interactively:
```python
import Tkinter
Tkinter._test()
```
It was shown animated dialog window with buttons, as expected, and I thought that it will be enough.

I propouse to resolve described issue by attached patch.
[0002-Values-of-argc-and-argv-transfers-to-python-environm.patch.txt](https://github.com/KiCad/kicad-winbuilder/files/2172804/0002-Values-of-argc-and-argv-transfers-to-python-environm.patch.txt)
It adds single line of code to kicad, that transfers values of argc and argv variables to python environment.

This patch is tested and works fine on kicad builded with last kicad-winbuilder:
```Application: kicad
Version: (5.0.0-rc3-dev-2-g101b68b61-dirty), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.59.0 OpenSSL/1.0.2o (WinSSL) zlib/1.2.11 brotli/1.0.4 libidn2/2.0.5 nghttp2/1.32.0
Platform: Windows 7 (build 7601, Service Pack 1), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.67.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.59.0
    Compiler: GCC 7.3.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```
P.S.
To run kicad i had to add libbrotlidec.dll and libbrotlicommon.dll to kicad's bin dir, and I actualy don't know is it my fault or some new dependece is missed.

nickoe, 2018-07-07:
@baranovskiykonstantin 

Ok, that does indeed seem to work for me too. But I guess I would like to see that merged in kicad before I add it to the installer.

nickoe, 2018-07-07:
@baranovskiykonstantin Ok, I also see the issues with libbrotli. It seems like a dependency has changed for curl, I have pushed an update for that.

About the second patch, I expect you send it to the developers list.

baranovskiykonstantin, 2018-07-08:
@nickoe, the patch was sent to KiCad devlist.

nickoe, 2018-07-08:
The email thread

https://www.mail-archive.com/kicad-developers@lists.launchpad.net/msg30982.html

PR #66 (2018-07-10): Tk is installed with KiCad (as part of python).
Fixes #60
commit 0ebc502b (2018-07-10): Tk is installed with KiCad (as part of python).

Fixes #60
closed 2018-07-10 by nickoe