KiCad/kicad-winbuilder #62: Possibilities for v4->v5 update

EeliK, 2018-06-25:
Installing v5 replacing v4 or alongside v4 will be a big issue in the near future. So will using old v4 projects with v5. I have wondered if the Windows installer could offer some help for that. The involved issues are difficult and complicated no matter who has to solve them - KiCad with its internals, the installer, or the end user. If someone isn't familiar with the issues, here are some experiences:

https://forum.kicad.info/t/opengl-canvas-update/11153/13
https://forum.kicad.info/t/up-versioning-issues-v4-nightlies-v5/11256

I tried to come up with some general explanation of possible alternatives here:
https://forum.kicad.info/t/thoughts-about-advice-for-upgrade-from-v4-to-v5/10895/7
Unfortunately that thread is in "lounge" category and I don't know who has access to it. If necessary, I'll copy the whole post here.

The reason I write this git issue is that it would be worthwhile to think if at least some of the problems could be solved with the installer. Would it be possible for example to offer several KiCad 5 launcher scripts, like "from scratch", "for v4 projects with old libraries" and "just continue from where you left v4".

I see two big issues when people try to update to v5 from v4. The first is the fp-lib-table. That was asked for already in another issue. The second is the environment variables. As I see the probable issues at the moment, everything revolves around those two things. The issue with fp-lib-table is quite clear. The environment variables may be more difficult.

The main problem with the variables is that if they have been set in the system (Windows registry) for v4, they will affect v5, too, and are wrong. And if they are changed in the system for v5, v4 doesn't work anymore. If they are unset before installing v5 (removed altogether from the system) and not replaced with v5, the new version will work but it's unclear to me whether v4 still works. And in any case, if the users want to use both v4 and v5 or use v5 for old projects, they have to do some complicated preparations with environment variables, libraries and fp-lib-table. These are shortly explained in the forum post which I mentioned above.

In general I think, at least at the moment, that setting environment variables for the system (or the user) is wrong and should be avoided. Even if it's done for a clean v5 install without old v4 installation, it will probably cause troubles for future updates. KiCad does well without system environment variables, the only one which is needed is KICAD_CONFIG_HOME. It can be changed for example in a wrapper script/batch file as I have described in https://docs.google.com/document/d/1Rq8i2Ay7qpGpffaj-AQmE-Xp88ikHhgyt0Ygpi8717o/edit?usp=sharing.

I don't have any ready-made or well thought solutions or suggestions for the installer, but this is food for thought. I'm ready to answer any questions if people are willing to discuss about this. Meanwhile two minimal things come to mind:

- Add a note about manually checking and updating fp-lib-table.
- Add documentation for the environment variables checkbox, preferably a warning against using it for clean installations. Maybe even put it into its own page, adding an option to clean up old variables.

EeliK, 2018-06-26:
Here's one new thread about the installation process:

https://forum.kicad.info/t/does-v5-have-to-overwrite-on-install/11282

dnotq, 2018-06-27:
Devs, would it do any good if I downloaded the source and worked on a pull-request for this?  Would it be considered / accepted?

marekr, 2018-07-08:
@nickoe Honestly nick, it would be cool if we could move to a new folder scheme.

C:\Program Files\KiCad\4.0
C:\Program Files\KiCad\5.0

future 
C:\Program Files\KiCad\6.0

Basically any CAD program installs like that (electrical and mechanical) not to mention bigger IDEs like Visual Studio or Clion as it allows you even have both installed side-by-side for compatibility reasons. Like you may want to use 5.0 for new designs but you want to use 4.0 to tweak old designs and not risk any issues migrating to 5.0 when you only have to tweak one thing.

But that's just one thought.

nickoe, 2018-07-08:
As the user config is not handled by the installer, but by kicad itself, I would expect this to be changed in kicad.

marekr, 2018-07-08:
@nickoe yea thats why I sad we have to tweak kicad. i.e. that email I started in dev chain.

nickoe, 2018-07-09:
For easy reference, I add it to the bug tracker as well

https://www.mail-archive.com/kicad-developers@lists.launchpad.net/msg30989.html