KiCad/kicad-winbuilder #77: ver 5.02 using KiCad ver 4 libraries for footprints ?

Aggebitter, 2019-01-09:
New to KiCad and trying to learn it on Windows. But noticed that "all" footprint associations is named wrong.

Look in the default folder "C:\Program Files\KiCad\share\kicad\modules"
KiCad is looking for default footprint "Resistors_ThroughHole.pretty "
The file is named "Resistor_THT.pretty

So for each faulty name I have to add "s" Resistor(s), Capacitor(s) and change THT to ThroughHole 
And that is on the majority of the footprint libs

I'm I doing wrong ?

nickoe, 2019-01-09:
Remove your old config of fp-lib-tables and reinitialize.

poeschlr, 2019-01-09:
A more detailed description of what is going on is found in the forum FAQ: https://forum.kicad.info/t/i-had-kicad-4-installed-previosly-now-i-updated-to-v5-now-i-have-some-problems-with-the-library-setup/11932