KiCad/kicad-winbuilder #79: Fails to Build

GyrosGeier, 2019-03-23:
The current state fails to build as it doesn't find the "Chinese" language support in NSIS. The NSIS package ships with "SimpChinese" and "TradChinese" language files.

nickoe, 2019-04-08:
FYI, we still need to try the suggestion from GyrosGeier

nickoe, 2019-05-20:
There were two changes need to make this work. First off including for example SimpChinese as suggested by @GyrosGeier and then also use the `LANG_SIMPCHINESE` for the kicad related translations.

https://github.com/KiCad/kicad-winbuilder/compare/2e51eb666fa47bc20f551fb38e9b5f4f31486eb3..55819fa484eb272fa1f7959a91ad21b27f8a4624

@taotieren Could you check if this works as expected on the latest windows nightlies?

I have only verified that it did build.

closed 2019-05-20 by nickoe

taotieren, 2019-05-21:
@nickoe  Okay, I'll try to build the latest Windows nightlies.

taotieren, 2019-05-22:
@nickoe  Using kicad-5.1-jenkins-281-x86_64.exe of test branch, I tried to build KiCad myself. When I generated mingw-w64-x86_64-kicad-git-r13143.1181b9f9b-1-any.pkg.tar.xz, I couldn't generate KiCad-XXX.exe script automatically without any error message and clueless, hoping to give some ideas for troubleshooting. Thank you.

nickoe, 2019-05-22:
@taotieren  I don't understand why you have issues. If you are to build yourself, it is easier to just run `make_x86_64.bat`.

Otherwise, just download https://jenkins.simonrichter.eu/job/windows-kicad-msys2-patch/357/artifact/out/pack-x86_64/kicad-patched-357-bc52f0a03-x86_64.exe and check that translations work in the installer.

taotieren, 2019-05-22:
Yes, I did, but it only generated *. pkg. XZ and did not generate *. exe.

taotieren, 2019-05-22:
I'll try to rebuild it. Maybe there's something wrong with the system environment on my side.

taotieren, 2019-05-22:
Can I compile the KiCad*.exe installation package under Linux?

nickoe, 2019-05-22:
In principle yes, but it is not supported.

nickoe, 2019-05-22:
What is the error you get?

taotieren, 2019-05-24:
`-- Installing: F:/git_clone/kicad/kicad-winbuilder/msys64/home/YI/MINGW-packages/mingw-w64-kicad-git/pkg/mingw-w64-i686-kicad-git/mingw32/share/kicad/template/pagelayout_logo.kicad_wks
==> 正在清理安装...
  -> 正在删除 libtool 文件...
  -> 正在清除不打算要的文件...
  -> 正在从二进制文件和库中清除不需要的系统符号...
  -> 正在压缩 man 及 info 文档...
==> 正在检查打包问题...
==> 正在创建软件包"mingw-w64-i686-kicad-git"...
  -> 正在生成 .PKGINFO 文件...
  -> 正在生成 .BUILDINFO 文件...
  -> 正在生成 .MTREE 文件...
 正在压缩软件包...
==> 完成创建：mingw-w64-i686-kicad-git r13164.75cea18f0-1 (2019年05月23日 19:41:13)
-- Success 0: cd "/f/git_clone/kicad/kicad-winbuilder/msys64/home/YI/MINGW-packages/mingw-w64-kicad-git" && export CARCH=x86_64 && TERM=vt220 makepkg-mingw -s --noconfirm

NI@NI MSYS /f/git_clone/kicad/kicad-winbuilder
`
Compiled under the corresponding directory：
`mingw-w64-i686-kicad-git-r13164.75cea18f0-1-any.pkg.tar.xz`

No corresponding compiled 
`mingw-w64-i686-kicad-git-r13164.75cea18f0-1-any.exe`

How to convert *any.pkg.tar.xz into *any.exe

nickoe, 2019-05-24:
@taotieren If you use the make_x86_64.bat script it should also call the copydlls.sh script which calls NSIS to create the installer. Keep in mind that the exe might be in $HOME/out inside the msys2 environment.

So in `/f/git_clone/kicad/kicad-winbuilder/msys64/home/YI/out`