KiCad/kicad-winbuilder #89: python3 build required

MitjaNemec, 2019-10-06:
As an action plugin writer, I'd like to get my hands on python3 enabled build for windows in order to test the backend (non-GUI) part of plugin code. @nickoe tried to make the build a couple of times (see: [Mitja Nemec's action plugin issue 36](https://github.com/MitjaNemec/Kicad_action_plugins/issues/36#issuecomment-522315562) ), but failed a couple of times.

nickoe, 2019-10-06:
@MitjaNemec Are you sure you were testing with the correct build? In the last one the python.exe is for python 3.

```
$ wine python.exe 
Python 3.7.1 (default, Oct 21 2018, 19:06:06)  [GCC 8.2.0 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

https://kicad-downloads.s3.cern.ch/windows/testing/patched/kicad-patched-502-8c2ee6c02-x86_64.exe

MitjaNemec, 2019-10-06:
This is really weird. I've reinstalled it and now I have 3.7.1. I really don't have a good explanation for this, but obviously the problem was on my side, and for this I'd like to apologize for the inconvenience I've caused.

As I have KiCad 5.1.4 under `C:\Program Files\KiCad` and I install python3 versions under `C:\Program Files\KiCad_py3`. In previous attempts I did not delete previous content before installing. But now I deleted the installation folder content before installing and it works. So it might be that the installer did not overwrite the interpreter. I'll check this on my work computer hopefully tomorrow. So I'll close the issue tomorrow when I'll post the findings what happened on my work computer. Thanks for the support.

MitjaNemec, 2019-10-06:
Obviously, I might have been checking in the wrong folder, but I've done this multiple times in the past and I doubt I'd make the same mistake every time, but it is possible.

nickoe, 2019-10-06:
The installer should overwrite, so I think it is more likely that you made a PEBCAK.

MitjaNemec, 2019-10-07:
Yes it was definitely PEBCAK. But I can confirm that it was only yesterday. As I had on work computer:
Version: (5.1.2-148-gf0ac61a8f), with KICAD_SCRIPTING_PYTHON3=ON and it reported python version 2.7.15

After installing kicad-patched-502-8c2ee6c02-x86_64.exe in the same folder without deleting the folder contents, the installer overwrote so now I have python version 3.7.1

Anyhow, thanks for the support. Much appreciated.

closed 2019-10-07 by MitjaNemec