KiCad/kicad-winbuilder #96: GitLab Winbuild migration

sethhillbrand, 2019-11-27:
Hi Winbuilders-

As part of the Great KiCad GitLab migration of 2019, I am testing the movement of open issues and PRs from GitHub to GitLab.

Here's what we know so far:

- Open Pull Requests will transfer to GitLab without moving the source of the PR.  That means pull requests will become GitLab Merge Requests that are referenced to the user's GitHub repository
- Open Issues will transfer to GitLab issues
- Closed/Resolved PRs and Issues will also transfer

In order to facilitate the transfer, it will be helpful if you can do two things:
- Open a GitLab account (these can use your GitHub OAuth)
- **Set your GitHub email to public**
- Let me know who the responsible parties are for the repository (@nickoe Anyone else?)

The second point is critical because the migration script will only link users whose e-mail addresses match.  If the script can't see your GitHub e-mail address, the issue will be tied to the KiCad-Bot that performs the migration.

The timeline for the actual migration will be determined by the responsible parties from bullet 3 above but it would greatly assist my testing for bugs if you could do the first two steps above as soon as possible.

Thank you-
Seth

nickoe, 2019-11-28:
@sethhillbrand there are no one else.

sethhillbrand, 2019-11-28:
@nickoe, Thanks!

closed 2019-11-28 by sethhillbrand